pibooth/pibooth #440: How can I modify the screen interface (remove option to print)?

subdrivr, 2023-03-18:
Wow, what a great application! Thanks for all the hard work on this project.  I am a total NOOB on the Raspberry PI and coding in Python so excuse my ignorance.  I have Pibooth up and running with GPIO buttons, even have plugins for Google picture storage and QR code installed and working great. My question is fairly simple, but I can't seem to figure out how to change the code.  I desire to run Pibooth as a picture upload app only, i.e. no option to print the picture. How can I change the following:
-On the "Print the Photo" screen I want to remove the words "Print the Photo?" and associated arrow (and turn off associated LED output) and just keep the words "Forget this photo" and associated arrow and blinking button.
-After the "Thanks" screen, I would like to remove the "You can still print this photo" and associated arrow (and turn of associated LED)  and just move on to the next picture capture process.
I've scrubbed the code and understand the basic hook and state structure, but am totally baffled on how the above words and arrows are generated on the screens or how to remove/change them.
Thanks!

anxuae, 2023-03-18:
In the config file, you can define the printer to use. The current value is 'default' if you never change it. That means pibooth will you the first find printer.

If you define a name of a fake printer, for instance 'noprinter': all arrows for print action will be disabled

subdrivr, 2023-03-19:
Perfect!  Exactly what I wanted.  Thanks so much for the quick reply!

closed 2023-03-19 by anxuae