pibooth/pibooth #442: Print with a receipt printer

biscuitbox, 2023-03-19:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I tried to print a photo captured with a receipt printer, but the quality is worse than I expected. Of course, there is a limit to the resolution since it is only 180dpi, but I wish there was an option to convert the captured photo to grayscale and adjust the brightness and contrast.

Nice job. Good luck~

anxuae, 2023-03-20:
Hi, there is no option in pibooth to do that because it is really specific to a use case 

However you can make a plugin to customize the factory used to generate pictures. You will be able to change image paramètres.

closed 2023-03-20 by anxuae

biscuitbox, 2023-03-20:
Thanks for your answer. I'm not good at coding yet, but I'll try my best to learn.