pibooth/pibooth #447: Wiring Help

Chiv-Lives, 2023-03-28:
I've been using this tutorial: https://pibooth.readthedocs.io/en/latest/sources/install.html#circuit-diagram
And also this tutorial:  https://www.hackster.io/sabat54i/wedding-photo-booth-with-raspberry-pi-1cea3a
I have the box built, the button attached, and the pibooth working, but with a keyboard and not the button. I only have 1 button.
I bought a breadboard kit and have the breadboard mostly set up as the first linked tutorial suggests, but I'm confused on the button portion. The first one tells me I need two buttons, but they're so small, no one can press them, and I'm not keeping the breadboard/pi outside of the box. That would look ugly. The second tutorial shows how the wiring for the button is meant to go, however, I don't know where on the button the jumpers are supposed to be. There is what looks like an American electrical plug, without the grounding.
I had looked at the Gallery portion of the pibooth page and instead of showing me other people's pibooths that I can reference, it showed me a new page I've never seen with various backgrounds, etc. Did the website/code change?
I'm needing suggestions on how to wire the button properly, as I can't return the box.
Here's an Imgur link to pictures of the mechanism and breadboard for reference. The button screws onto the top of the mechanism.

> <blockquote class="imgur-embed-pub" lang="en" data-id="a/OV5hpQJ"  ><a href="//imgur.com/a/OV5hpQJ">breadboard and button</a></blockquote><script async src="//s.imgur.com/min/embed.js" charset="utf-8"></script>

srvoets, 2023-03-28:
I'm assuming its a momentary button.  I was new to this not that long ago.

Buttons need a positive and a ground.  Positive goes to an open GPIO Pin on the pi, and ground, to ground.  Pressing it will temporarily close the circuit, and boom button press is registered.

If it is an arcade style button that lights up, you need to wire up the button as above, and then also connect power and ground to the remaining two prong to make it light up.

I know this is not a diagram, but for a while, diagrams didnt make sense to me, maybe that explanation helped.

Chiv-Lives, 2023-03-29:
> I'm assuming its a momentary button. I was new to this not that long ago.
>
> Buttons need a positive and a ground. Positive goes to an open GPIO Pin on the pi, and ground, to ground. Pressing it will [...]

How do I wire up the button? Like, do I take the jumper wires and put them through the holes and manipulate them to stay in place? Also, it is a light up button. Do I _HAVE_ to put power to it, or can I just let it be not lit up? I had tried to use it yesterday with the wires where I thought they should be, gave it a test, and nothing happened.
There are a few possibilities as to why it didn't work;
1. I wired it wrong
2. I've killed the wires, either by bending the metal too much, or breaking the connection
3. the button isn't programmed into the PiBooth software yet.

I've put the Always On in peg 16 and ground into 14, which I've seen are the correct placements.

Chiv-Lives, 2023-03-29:
> I'm assuming its a momentary button. I was new to this not that long ago.
>
> Buttons need a positive and a ground. Positive goes to an open GPIO Pin on the pi, and ground, to ground. Pressing it will [...]

Okay, so I got it working. 
I was able to get the jumpers wired onto the button by bending the receiver (the long bit that should plug into something) through the holes on the button. I had plugged them into peg 5 and 6 on the RaspPi, like the one tutorial suggests. It didn't work because in the settings (Pibooth/config) the button press was mapped for 11, not 5. Once I switched it, the button functioned correctly.

closed 2023-03-31 by Chiv-Lives