pibooth/pibooth #448: Open CV keeps crashing when taking multiple photos

kadiepi, 2023-03-31:
**Describe the bug**
Pibooth keeps crashing when I try to take multiple photos in a row. If I choose "one photo", it works successfully. If I then go to choose "one photo", the app crashes. If upon start up I choose "4 photos", the app also crashes. It seems to be an issue with Open CV but I don't know how to fix it.
I always get the VIDEO IO ERROR: V4L2: Pixel format of incoming image unsupported by OpenCV

**Steps to reproduce the bug**
1. Open pibooth
2. Click on "take 4 photos"
3. See error.

**Logs**
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ DEBUG   ] pibooth           : Preview resolution is (640.0, 480.0)
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : took 283.071 seconds
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : took 1.227 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (4 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[... 182 lines cut ...]
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/pibooth/pibooth/plugins/camera_plugin.py", line 103, in state_capture_do
    app.camera.capture(effect)
  File "/home/pi/pibooth/pibooth/camera/opencv.py", line 220, in capture
    raise IOError("Can not capture frame")
OSError: Can not capture frame
[ DEBUG   ] pibooth           : took 0.121 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.103-v7+', version='#1529 SMP Tue Mar 8 12:21:37 GMT 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.057 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'

**Environment**
 - Raspbian Buster
 - Logitech Pro 5000
 - Pygame 2.3.0
 - Python 3.7.3
 - pygame meny 4.0.7

werdeil, 2023-03-31:
Hi @kadiepi Do you confirm that you are using effects on your captures? I would say you have this bug https://github.com/pibooth/pibooth/issues/425 which will be fixed in next release.
To confirm you can try to remove the effects in your config.

anxuae, 2023-04-08:
@kadiepi can you confirm that you are using effect?

kadiepi, 2023-04-08:
Hi, I am not using any effects. This is straight out of the box.

anxuae, 2023-04-08:
This error may also append when the resolution specified in your config is not supported by your webcamera

gilou, 2023-04-12:
I can confirm the issue on v2 and v3, first capture goes fine, the next one does not… Then I read the comment about the resolution, set to (640, 480)... buuuuut ! `Putting preview resolution back to (1280.0, 720.0)` 
So I set the resolution to just that, and it stops crashing. But that what makes it unhappy..

`self._preview_resolution = (self._cam.get(cv2.CAP_PROP_FRAME_WIDTH), self._cam.get(cv2.CAP_PROP_FRAME_HEIGHT))`
Maybe that could be set to `min(resolution[0], cv2.CAP_PROP_FRAME_WIDTH)` or what not to avoid trying to change it to something it does not like… 

Or maybe, let's just make sure we set the resolution to the max capabilities ;)

anxuae, 2023-04-13:
Hi @gilou 

Thanks for repporting. the line `self._preview_resolution = (self._cam.get(cv2.CAP_PROP_FRAME_WIDTH), self._cam.get(cv2.CAP_PROP_FRAME_HEIGHT))` is used to save the default webcam resolution which is generally the one optimized for best video.

This is the user responsibility to pus a working resolution in the config

gilou, 2023-04-13:
agreed, and it's fine by me!
There might be a way to improve the OpenCV / V4L interaction to avoid the crashes though, even on a supported resolution however. I only use gphoto2 on actual booths, so I'm not too interested (and I made pibooth-stillimage-camera to work around it anyway ;))

kadiepi, 2023-04-13:
On my end, I changed the resolution to the preview resolution and it still didn’t work. I even tried finding the compatible resolutions and changed the settings, but all of these result in a crash after taking more than one photo