pibooth/pibooth #449: One button control

ChinaeduO, 2023-04-07:
I'm frustrated. I build a photobooth with one big buzzer. I would like to control pibooth with just one button. With short press and long press.

The software should count the time the button was held:
If the user presses the button in the choose state for e.g. more than 2 seconds, then a collage is selected. If he releases it before, so < 2 seconds, just one single photo mode is selected.

I have tried to add such a feature, but I don't know where to add it, so do you have any hints?

anxuae, 2023-04-08:
Hi @ChinaeduO,

This is not easy to achieve, you can try to modify this tow functions:

https://github.com/pibooth/pibooth/blob/19851a161af6a09f8e5554d8ec2eb7194c980972/pibooth/booth.py#L197-L248

ChinaeduO, 2023-04-08:
I modified the code a bit to capture how long the button is pressed:

```
def _on_button_capture_held(self):
    start = time()
    if all(self.buttons.value):
        self.buttons.capture.hold_repeat = True
        if self._multipress_timer.elapsed() == 0:
            self._multipress_timer.start()
        if self._multipress_timer.is_timeout():

                # Capture was held while printer was pressed

            if self._menu and self._menu.is_shown():

                    # Convert HW button events to keyboard events for menu

                event = self._menu.create_back_event()
                LOGGER.debug('BUTTONDOWN: generate MENU-ESC event')
            else:
                event = pygame.event.Event(BUTTONDOWN, capture=1,
                        printer=1, button=self.buttons)
                LOGGER.debug('BUTTONDOWN: generate DOUBLE buttons event'
                             )
            self.buttons.capture.hold_repeat = False
            self._multipress_timer.reset()
            pygame.event.post(event)
    else:

            # Capture was held but printer not pressed

        if self._menu and self._menu.is_shown():

                # Convert HW button events to keyboard events for menu

            event = self._menu.create_next_event()
            LOGGER.debug('BUTTONDOWN: generate MENU-NEXT event')
        else:
            LOGGER.debug('Perssed')
            while self.buttons.capture.is_pressed:
                LOGGER.debug('Holding')
            length = time() - start
            LOGGER.debug('Holded for: ' + str(length))

                # event = pygame.event.Event(BUTTONDOWN, capture=1, printer=0,
                #                           button=self.buttons.capture)

            LOGGER.debug('BUTTONDOWN: generate CAPTURE button event')
        self.buttons.capture.hold_repeat = False
        self._multipress_timer.reset()

            # pygame.event.post(event)
```

It is working, I can see the debug logs when the button is pressed. But, however, although I am constantly pressing the button, something is causing a bounce event. In the logs, you can see that it hops out of the while loop, although the button is still pressed physically. I tested it in an external python script and there it seems to do just fine. Only in context with the pibooth software, this weird behavior is happening.

        
```
[ DEBUG   ] pibooth           : |  Perssed
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[... 126 lines cut ...]
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holded for: 0.3837904930114746
[ DEBUG   ] pibooth           : |  BUTTONDOWN: generate CAPTURE button event
[ DEBUG   ] pibooth           : |  Perssed
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holding
[ DEBUG   ] pibooth           : |  Holded for: 0.12746906280517578
[ DEBUG   ] pibooth           : |  BUTTONDOWN: generate CAPTURE button event

```

Maybe somewhere in the code, the button is forced to be inactive again?

anxuae, 2023-04-08:
It's normal, because long-press is already used to open menu.
So pibooth use the delay set in `[CONTROLS][multi_press_delay]` to monitor if both buttons `capture` + `printer` are pressed together.
After this delay, if `printer` button is not pressed, the capture is triggered

ChinaeduO, 2023-04-08:
Noah, even if I am setting the `[CONTROLS][multi_press_delay]` to unreachable high, or even if I comment out the block completely that handles the multipress logic, the holding of the button is interrupted, although it is physically still pressed.

anxuae, 2023-04-08:
Maybe a bug in the `gpiozero` `hold-repeat` function. Pibooth don't use "pressed" event, but "held" event (necessary because menu is shown after a long held time)

Can you provide you simple script?