pibooth/pibooth #450: Hardware buttons bugged

sjrgrd, 2023-04-07:
**Describe the bug**
Build a photobooth two months ago. Been working great, used it for several occasions, never failed me. Last time it was used was April 2nd 2023.
Booted it up today and found the hardware buttons did not do as expected. Pressing the capture button a first time does nothing. Pressing it again within a few seconds enters the menu (as if I pressed ESC on the keyboard). Another press does nothing but pressing it again within a few seconds exits the menu again.
Longpress also enters/exits the menu.
If I wait more than ~3 seconds before second press nothing happens.
Same is true if I enter the "select layout"-screen by pressing P on the keyboard.
The LEDs in each button is blinking relevant to what state pibooth is in (only "capture" LED blinks in waiting state; both blinks at "select layout" state).
I notice the logs stating anything from 10 to 350 seconds between button presses even though it was only a few seconds.
Everything works perfect using the keyboard.

I went over the wiring a dozen times to check for bad connections. All good.
I reinstalled the OS and installed pibooth from scratch using the docs. Same issue.
I replaced the physical buttons with another pair. Same issue.

I just noticed pibooth was updated on April 2nd, could this be a bug in the new release?

**Logs**
The mystery is the BUTTONDOWN actions seen below. This logs is from a long press on the button:

[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : |  BUTTONDOWN: generate DOUBLE buttons event
[ DEBUG   ] pibooth           : |  BUTTONDOWN: generate MENU-ESC event
[ DEBUG   ] pibooth           : +> took 350.164 seconds
[ DEBUG   ] pibooth           : +< Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756

**Environment**
 - OS : Raspberry Pi OS Buster
 - Camera : Canon EOS 1000D
 - Pibooth version 2.0.7

anxuae, 2023-04-08:
Hi @sjrgrd,

> Booted it up today and found the hardware buttons did not do as expected. 

As I understand, you boot it without any update and it does not work. In this case, the new version is unrelated to your issue

sjrgrd, 2023-04-08:
> Hi @sjrgrd,
> 
> > Booted it up today and found the hardware buttons did not do as expected.
> 
> As I understand, you boot it without any update and it does not work. In this case, the new version is unrelated to your issue

To be honest I’m not sure if I ran a ‘sudo apt update/upgrade’ before shutting it down April 2nd. 

How can I install v2.0.6 to test it out?

anxuae, 2023-04-08:
`sudo apt update/upgrade` does not update pibooth.

But you can install a previous version using `sudo pip3 install pibooth[dslr,printer]==2.0.6`

sjrgrd, 2023-04-08:
Then it’s most likely unrelated to pibooth. I will test with 2.0.6 again asap and if problem persist we can close this issue.

sjrgrd, 2023-04-08:
Removed 2.0.7 and installed 2.0.6, same issue. 
Tested with other GPIO pins, same issue. The LEDs are working properly. 
I have no idea what to do from here on. Could test with my other pi4 currently in the arcade. 
Cannot find any info on changes to the GPIO pinout, so not sure if it’s an OS issue?
I’m closing this issue, seems unrelated to pibooth. 
Thank you for the wonderful software!

closed 2023-04-08 by sjrgrd

werdeil, 2023-04-08:
@sjrgrd Even if it is not related to pibooth code, the behavior you get is showing that the "print" button is always considered as pressed (pressing capture is seen as a double press). Did you test both while checking wiring and GPIO pinout?

sjrgrd, 2023-04-08:
> @sjrgrd Even if it is not related to pibooth code, the behavior you get is showing that the "print" button is always considered as pressed (pressing capture is seen as a double press). Did you test both while checking wiring and GPIO pinout?

Same issue even though I completely remove the print button. Also tried connecting pin 11 to ground with a small wire (to simulate button press) to make sure it’s not a problem with the switch.

sjrgrd, 2023-04-08:
This is after double press or long press:
`[ DEBUG ] pibooth : | BUTTONDOWN: generate DOUBLE buttons event`

This is in the menu after double press or long press:
`[ DEBUG ] pibooth : | BUTTONDOWN: generate MENU-ESC event `

sjrgrd, 2023-07-26:
I ended up replacing the Pi 4 but using the same SSD and the problem was gone.
Conclusion: it was a hardware failure and not related to the software.

mmourcia, 2023-12-31:
Hi there,

I met the same problem with 2 differents pi 4.
I first thought it was a hardware problem and when I replaced the first pi by a second, buttons worked again. 
And one day, it began to fail on the second pi too.
I really hope it's not another hardware problem because I can't understand why it happens.
Both pi were used only with pibooth software.

sjrgrd, 2024-01-01:
> Hi there,
>
> I met the same problem with 2 differents pi 4. I first thought it was a hardware problem and when I replaced the first pi by a second, buttons worked again. And one day, it began to fail [...]

Thank you for commenting.
I haven't had the chance to troubleshoot the pins on the failing pi4. Would be interesting to test the pins with a debug tool to see if it is indeed a hardware failure.
The second pi4 i installed has been working since summer 2023 without any problems. Been using the booth on several occasions.

NiraCity, 2024-04-10:
Hello, I have the same problem.

The above message corresponds exactly to my situation.

The picture_btn worked well in the beginning, but unfortunately not anymore. (less than a week)
Now it only works if you hold it down for a long time, then the menu will open.
Even chaneg the input to another pin wil not solve this problem.

I use a relay board to switch some outputs. I had to adjust some outputs in the setup (set them to a different pin number), but it worked well for a while.

Anyone have a idea? 
Do I need to purchase a new RPI and how do I arrange this for the future?

werdeil, 2024-04-10:
Hi @NiraCity, from my understanding of the issue this is linked to the way your hardware buttons are wired; It seems that even not pressed the voltage is still low enough to consider it as pressed.

Do you have tried to add a resistor in the circuit hardware button, you can search for wiring example on the internet.

NiraCity, 2024-04-10:
Oke thanks, I will try the solution you sent in the picture.
In the manual you switch 0v and in the example 3,3v.

It was working with the original drawing, switching the 0v to the selected pin.

But i will try this later. Thank you

DJ-Dingo, 2024-04-10:
Yes, use what it says in the manual. I cant remember if you should use pull-ups or pull-downs.

But again that is why i mention both types "make a Internal pull-ups or a pull-down to the button".  I am not home so can´t look at mine, "if it is Pull-Up or Pull-Down". 

Rpi do have internal Pull-Up/Pull-Down resistors, but i have always used the physical ones as well, and can´t remmenber what Pibooth do in the programing, if it use it or not. 

Yes that is a good video to show both methods. And it is a verry simpel thing to do. 
> This is a good video about switches https://youtu.be/Bqk6M_XdIC0?si=dj10kpk-dxE0XoKz

NiraCity, 2024-04-12:
I just received the resistors. I connected everything as above, but no result.

Then I started playing around with the push button times.
I have the value of,
# How long to press multiple hardware buttons in seconds
multi_press_delay = set to 25 seconds.

Now when I press the button **nothing** happens. 
If I wait 25 seconds, it opens the menu, so the input seems to be correct and activated for 25 seconds.
It seems like the push button funcion is turned off. 
Maybe I turned this off unconsciously? 

I have also changed the type of screen so that.

NiraCity, 2024-04-15:
The problem is solved.

I had mounted the resistors as shown in the picture. 
Unfortunately it didn't work. 
Then I realized that I had not connected the print button. 
afterwards that input was floating this changed the funktion of the push button "take picture"  to "go to menu".

I had any problems with it in the first two weeks, so I didn't think that was the fault.

Thank you for your help.