pibooth/pibooth #451: Pictures show up as black

kadiepi, 2023-04-08:
**Describe the bug**
I am using a Logitech C270 webcam. I was able to test the webcam separately and the images save perfectly fine, however when I take any images with pibooth, the resulting image (in the raw folder and the Pictures folder) is an entirely black square. During the picture taking process, the picture shows up fine as a preview, so I don’t know why it’s saving as black.

**Steps to reproduce the bug**
1. Take any number of pictures (previews look good)
2. During image processing, final results is a black square

**Logs**

**Environment**
 - OS Raspbian Buster
 - Camera Logitech c270
 - Pygame 2.3.0 python 3.7.3
 - Pygame menu 4.0.7
 -

anxuae, 2023-04-08:
What I'm seeing is that you have a video error:

```
VIDEOIO ERROR: V4L2: setting property #30 is not supported
```

I don't have Logitech C270 webcam, so difficult to help you. Maybe the community may have an idea.

werdeil, 2023-04-10:
I have a Logitech C270 but I didn't used it on the Raspberry Pi, I'll try to change my setup to see if I reproduce.