pibooth/pibooth #453: Introducing pibooth-stillimage-camera to test/showcase pibooth

gilou, 2023-04-09:
Hi !

I'm exploring, and might be interested in using pibooth on some of my booths, that have custom apps that could benefit from all the amazing work done on pibooth!

To get my hands on the project, and to get a grasp on the plugin structure, I wrote pibooth-stillimage-camera as a plugin (though it could be integrated in the "core"). I'm more of a gitlab guy, so it lives here for now: https://gitlab.com/wolface/wolfoto-project/pibooth-stillimage-camera but could be hosted on GH if needed.

So the doc is quite straightforward (or at least I hope), and someone could simply use `pip install pibooth-stillimage-camera` and enable debug in the GENERAL  section of the conf, and just start pibooth without having an actual gphoto / rpi camera. That makes it very easy to go through the interface!

Also it can be used to tests more stuff, but you may already have a way to bypass that in the tests, I haven't read that deeply into the CI setup ;)

While doing that, I could get a nice idea about a few things in the core of pibooth, that may lead to contribution, or sponsored improvements to come…
- some code is hidden in GpCamera / RpiCamera, where it really could live in BaseCamera (that may have been imagined as a bit too abstract as well)
-  I'm not sure why the camera_proxy is heavily wired in BaseCamera either, even if we can set it to None, I feel the pattern is a bit weird, but that may be because I lack vision on the overall project!
- the "API" (esp. the Camera behavior) is not documented and I had to dive in the code to understand many interactions with pygame. The hook doc is great, but doesn't tell all the story on some parts!

My next step may or may not be enabling the use of a "classical" webcam using v4l2 as the camera, through another similar plugin ;)
Also it was quite great to dive in the code, I think it's quite nicely written, and easy to grasp, and some parts are quite elegant: well done!

Cheers,
Gilou

gilou, 2023-04-09:
Ah, another thing concerning the overall structure… Since effects in GpCamera are mostly PIL filters, that may be nice to have in the BaseCamera or easily available for other camera types ;)

werdeil, 2023-04-10:
Hi @gilou, thanks for the feedbacks and the test plugin ! No need to host it on GitHub, as you published it on pip it is already visible for everyone (https://pypi.org/search/?q=pibooth). We didn't had the need of this kind of hack yet as Pibooth can use the webcam of our dev computers when we are not on the Raspberry pi, but it is always nice to have a mock like this.

Thanks also for your comments on the code, but if you want to dive deeply in it my best advice would be to start the the [3.x](https://github.com/pibooth/pibooth/tree/3.x) branch. it is still in progress but it contains most of the new code that could answer some of your needs (or move some of the code). It has been done mostly by @anxuae but I can always help if he is not available. I see that you opened another issue with more specific questions I let you close this one if you want.

Merci beaucoup !

closed 2023-04-10 by gilou

gilou, 2023-04-10:
And it's quite amazing, most of what I was complaining is in fact adapted and improved in v3.x that I didn't see at first… Now we have a featureful BaseCamera, and it's a lot easier to access actable properties to interact! Awesome work, again!