pibooth/pibooth #456: Pibooth with a screen in "portrait mode"

Aurel-Charles, 2023-04-11:
**Describe the bug**
I want to use pibooth with a screen in portrait mode and fullscreen. But the app doesn't really behave like it see the new resolution of the screen and so half of the window app is crop on the right.
I've tried puting 1080X1900 in the config file for the window parameter. It kind of works but I'll still have to deal with the raspberry menu bar. Also during the capture time, it's still cropping the preview. so not really a solution here

**Steps to reproduce the bug**
1. change the raspberry configuration screen to have it in portrait mode
2. launch pibooth in fullscreen

**Environment**
 - Raspbian Buster
 - Rpi camera v2
 - Pibooth version 2.0.7

It looks like a similar issue #223 that have been already resolve but maybe I need a little help on this one

Thanks as always! Your work and all the updates are really good!

dafunkychild, 2023-08-22:
Hello,
It is the first time I encounter the same problem. It is also the first time I use a RPI.

Environment 

- RPI4
- Raspberry Pi OS ( release date 2023-05-03 )
- DSLR canon 1200D
- Pibooth version 2.0.8

noelg, 2024-02-22:
I had the same problem, and after some debugging I figured out it was an issue with pygame (pygame 1.9.6 is installed by default with pibooth).

I upgraded to pygame 2.5.2 (pip3 install pygame=2.5.2) and it works like a charm now.