pibooth/pibooth #459: Pibooth randomly takes picture without action of user

Refurb, 2023-04-13:
I have my Raspberry running with PiBooth. When I just keep it on and not doing anything on it for a while, it sometimes just gets a automatical trigger and starts to capture photos. Is this an option I can disable? I can imagine that the developers enabled this because the photobooth will get attention from people around it. But the camera will flash when this is happening without any reason and that's not really my wish.

I'm running the latest PiBooth version, a touch screen and a Canon EOS 1100D. I don't have a physical button but using the touchschreen as a "button". The printer is not attached when this is happening.

Many thanks in advance!

anxuae, 2023-04-17:
Hi @Refurb there this no such feature in pibooth. Are you sure that you don't have a key blocked which could trigger an event of a GPIO pin connected?

I can not reproduce it.

DJ-Dingo, 2023-04-17:
Hey @Refurb 
Usually it is disturbances on the GPIO (This can come from many things e.g. the Screen, Powersupply, Mobilephones, Camera, Cables for e.g. lights that are not properly shielded, etc.). But it can often be fixed by changing a few small things in the config file pibooth.cfg

You can test if this will help.
If you scroll down to CONTROLS inside the pibooth.cfg file and change "single hardware button", debounce_delay = 0.3 to say 0.8 or 1.0 (This is normally for a button, but since you are not using a button but a touch screen it will still be active on GPIO).

At the same time, you can change "Physical GPIO IN pin" picture_btn_pin = 11 to another pin that is not so close to your "Physical GPIO OUT pin to light". 

That way, you'll be lucky that they don't disturb each other. Set e.g. "Physical GPIO IN pin" picture_btn_pin, to e.g. 14, 17, 21 or 26 They are all far from PIN 7 to light "If you use standard PIN 7 for light". You can see all GPIO here https://pinout.xyz/

Since I don't know what else you use for GPIO PINs, it may be difficult to come up with other suggestions.

Here are the info from the standard config.cfg

[CONTROLS]
# How long to press a single hardware button in seconds
debounce_delay = 0.3

# Physical GPIO IN pin to take a picture
picture_btn_pin = 11

# Physical GPIO OUT pin to light a LED when the picture button is pressed
picture_led_pin = 7

closed 2023-04-24 by anxuae