pibooth/pibooth #461: Can not capture frame

davissimo, 2023-04-14:
Hi,

I did a fresh install of Raspberry Pi OS 64bit and PiBooth with default settings, but whichever camera device I use keep getting "can not capture frame"

Hercules webcam, as well an HDMI video capture device acting as a webcam:
A preview is displayed, and after the countdown:
```
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (961) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1214) setProperty OpenCV | GStreamer warning: GStreamer: unhandled property
[ INFO    ] pibooth           : Connected to printer 'Cat_Printer'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (4 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (536) startPipeline OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1085) setProperty OpenCV | GStreamer warning: no pipeline
[ ERROR   ] pibooth           : Can not capture frame
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='6.1.21-v8+', version='#1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023', machine='aarch64', processor=''
```
Sony A6000 mirrorless camera:
No preview is displayed, the shutter of the camera is triggered, then:
```
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Cannot identify device '/dev/video0'.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ INFO    ] pibooth           : Connected to printer 'Cat_Printer'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : Can not get camera preview image
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='6.1.21-v8+', version='#1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023', machine='aarch64', processor=''
```

Ideas?

anxuae, 2023-04-14:
Hi @davissimo , for the moment pibooth is not compatible with PiOS Bulleyes. Please install PiOS Buster

anxuae, 2023-04-24:
Hi @davissimo seems that OpenCV (library used for webcam) does not handle libcamera (new camera driver included in PiOS Bullseye)

Re-enable the legacy camera should solve the issue

Some info here: https://spyjetson.blogspot.com/2022/03/opencv-camera-control-on-raspberry-pi.html

closed 2023-04-24 by anxuae