pibooth/pibooth #464: pibooth-picture-template from pibooth version 2.0.6 to 2.0.7

dafunkychild, 2023-04-26:
I created my own template with  Flowchart Maker to use with 1 or 3 photos. It works fine on pibooth version 2.0.6.
The same template in pibooth version 2.0.7 works great if I select 1 photo but if I choose 3 photos, they stay in the default position and don't match with my overlay.

I noticed hybrid mode natively works on the computer I use and pibooth version 2.0.7. It was not working in version 2.0.6

**Logs**
2. Post your config file : [diagnostic.txt](https://github.com/pibooth/pibooth/files/11334471/diagnostic.txt)
3. Copy-paste all the logs from the console.  can't "There was an error creating your Issue: body is too long (maximum is 65536 characters).", so I attach the file.
[console.pibooth.txt](https://github.com/pibooth/pibooth/files/11334512/console.pibooth.txt)
4. Post the `diagnostic.log` file if you are using a DSLR camera (generated by running the command `pibooth-diag`)
[pibooth.txt](https://github.com/pibooth/pibooth/files/11334480/pibooth.txt)

**Environment**
 - OS [Ubuntu-22.04.2] I don't use a raspberry but a Chromebook Asus C434T
 - Camera Canon 1200D/
 - Pibooth version 2.0.7

dafunkychild, 2023-04-26:
I assume the template file could be helpful.

file:///home/simon/.config/pibooth/picture_template.02.xml.txt
0:-33:334:60

anxuae, 2023-04-28:
Hi @dafunkychild 

Thanks for your clear report.
I've tried the v2.0.7 with your template, and its working fine. Here is the picture generated:

[image]

Are you sure that you have restarted pibooth after your change is done?
Also be sure to put it in the correct path as defined in your config

dafunkychild, 2023-04-29:
Hey, thank you for your feed back. It means there is no issue with the template file.  Version 2.0.7 was a fresh install on a different computer.

The correcth path is set in the config , 1 photo match the overlay, but three photos don't.

anxuae, 2023-05-06:
Hi @dafunkychild 

I'm closing this issue for now. Please reopen if necessary.

closed 2023-05-06 by anxuae

dafunkychild, 2023-05-13:
I use a live persistent usb ubuntu distribution so I will reinstall everything and let you know.

dafunkychild, 2023-07-13:
So far I didn' t have a chance to reinstall but I finally did it today. This time using a classic unbuntu distibution on a SSD hard drive. 

There is no more problem with the template.