pibooth/pibooth #465: Switch off preview

Steff1990, 2023-04-30:
I have a general problem. I want to switch off the preview. Only want a countdown, no pictures. But everything i tried leads into an error. Tried to switch off previewtime to 0 and so an. 
Hope someone can help me.
Thanks

anxuae, 2023-05-06:
Hi @Steff1990 

A bug has been fixesd for futures version, to hide the preview by setting `[WINDOW]preview_delay` to 0.
But in that case the countdown is not shown.

There is now possibility to hide the preview and showing count down.

Concerning your error with hybrid mode, I nee the different logs as defined in the bug report template to understand what happened.

F3rlin, 2023-08-01:
Hi,

I have a similar issue. I want to turn of the preview completely. I have tried to set "[WINDOW]preview_delay to 0" but then pibooth crashes. In the menu there is no 0 option only 1 to 20.

is there another way to turn of the preview?

Thanks