pibooth/pibooth #466: Canon SX620 | Could not capture image | [-110] I/O in progress

Jordanracco, 2023-04-30:
**Describe the bug**

With my Canon SX620HS, when I try to take a capture
an error appears

The photo is successfully taken and saved in the SD card.

Preview works.

Thanks for your help

**Steps to reproduce the bug**

When try to take a picture

**Logs**

[pibooth.log](https://github.com/pibooth/pibooth/files/11361133/pibooth.log)
 file

[diagnostic.log](https://github.com/pibooth/pibooth/files/11361121/diagnostic.log)

**Environment**
 - RPI OS
 - Canon SX620HS
 - Pibooth version 2.0.7
 - GPhoto2 2.5.27

anxuae, 2023-05-02:
Hi @Jordanracco be sure to disable auto-focus drive on your DSLR
The auto-focus generally fails because it is too long, and Gphoto library generate an error "IO In process" 

In the diagnostic log, I found you are in "Auto"

```
/capturesettings/autoexposuremodedial
  Label       : Canon Auto Exposure Mode Dial
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Radio button
  Current     : Auto
  Choices     : ['P', 'TV', 'AV', 'Manual', 'Bulb', 'A_DEP', 'DEP', 'Custom', 'Lock', 'Green', 'Night Portrait', 'Sports', 'Portrait', 'Landscape', 'Closeup', 'Flash Off', 'Auto', 'Handheld Night Scene', 'HDR Backlight Control', 'Food', 'Grainy B/W', 'Soft focus', 'Toy camera effect', 'Fish-eye effect', 'Water painting effect', 'Miniature effect', 'HDR art standard', 'HDR art vivid', 'HDR art bold', 'HDR art embossed', 'Panning', 'Self Portrait', 'Hybrid Auto', 'Smooth skin', 'Fv']
```

closed 2023-05-02 by anxuae

Jordanracco, 2023-05-02:
Thank you for your reply @anxuae .

I can't change the mode with the command
"--set-config autoexposuremodedial=3"

Any choice returns an error. (0,1,2,3,..)
And the Canon turn off..

"gphoto2 --set-config autoexposuremodedial=3 --debug" :
[log2.txt](https://github.com/pibooth/pibooth/files/11374414/log2.txt)

Thanks for your help.

anxuae, 2023-05-02:
You have to turn in manually on your DSLR

Jordanracco, 2023-05-02:
The DSLR shuts down when I run the --set-config command.

When I run the command, it is "on" and detected with --auto-detect.
It turns off after the error :
*** Erreur ***
Échec de définition de la nouvelle valeur de configuration Manuel pour l'entrée de configuration autoexposuremodedial.
(Failed to set new manual configuration value for autoexposuremodedial configuration entry)
*** Erreur (-7 : « Problème d'E/S ») ***