pibooth/pibooth #468: Setup a slideshow on a second raspberry via wifi

dominiquegrota, 2023-05-04:
Hi @anxuae, It took me a few hours, but I got it ! the step by step is ready. 
The steps are : 
1.Install pipresents on a specific raspberry
2. Setup pipresents
3. Connect ssh between pibooth raspberry and pipresents rapsberry
4. create a rsync script
5. setup a plugin to fire rsync at each new picture on pibooth

Have fun 
Dominique

All steps are detailed here ->

install PIos legacy (debian burster) : https://www.raspberrypi.com/software/
parameters : add ssh + define user (pi + Password) + wifi (ssid + password)
Write the sd card
Insert the sd card in raspberry
The sd card is resized and reboots

set the country language and timezone at startup

Update Linux

     sudo apt-get update
     sudo apt-get upgrade

Install required packages

     sudo apt install python3-pil.imagetk
     sudo apt install unclutter
     sudo apt install mplayer
     sudo apt install python3-pexpect
     sudo pip3 install python-vlc
     sudo pip3 install selenium
     sudo apt install chromium-driver
	 
install Pipresents beep 

     sudo wget https://github.com/KenT2/pipresents-beep/tarball/master -O - | tar xz     # -O is a capital Ohhh...     wget https://github.com/KenT2/pipresents-beep/tarball/master -O - | tar xz     # -O is a capital Ohhh...	 
	 
seach the exact folder name 

	 ls -lia

Rename to "pipresents" the folder (should be named KenT2-pipresents-beep .....)

	 sudo mv  KenT2-pipresents-beep-xxxxxxxxxxx/ pipresents/

Test the pipresents install

	 python3 /home/pi/pipresents/pipresents.py

You will see a window with an error message which is because you have no profiles.Click OK to exit Pi Presents.

Get example files

     wget https://github.com/KenT2/pipresents-beep-examples/tarball/master /home/pi -O - | tar xz
	 
Move the content of the folder in /home/pi

	 cd  KenT2-pipresents-beep-examples-xxxxxxxxxxx 
	 mv * /home/pi

remove empty folder

	 cd ..
	 sudo rm -R KenT2-pipresents-beep-examples-xxxxxxxxxx
	 
Run pibooth liveshow

     python3 /home/pi/pipresents/pipresents.py -p pp_mediashow_1p4 -f -b

You should see a presentation.
     alt + F4 #to quit

Personalise the show : 

	 sudo nano /home/pi/pp_home/pp_profiles/pp_liveshow_1p4/pp_showlist.json
	 
You should modify these existing lines in the file: 
Delete default background image : 
   "background-image": "",
Set image size to full screen(fit = resize) x=0 y=0 window with=1920 widow height=1080 
   "image-window": "fit 0 0 1920 1080",
pictures should be ordered reverse = last picture first (the author added specially this option for our usage <3)
   "sequence": "reverse",
delete the default text
   "show-text": "",

OR use the grapic editor of show personalisation to edit /home/pi/pp_home/pp_profiles/pp_liveshow_1p4 : execute (graphic desktop version) /home/pi/pipresents the file Pi Presents Editor

run pipresents at startup
setup boot to desktop : 

	sudo raspi-config

  system option menu
  Boot / Auto login menu
  Desktop Autologin menu

create directory and copy autostart file  

	mkdir /home/pi/.config/lxsession
	mkdir /home/pi/.config/lxsession/LXDE-pi
	cp /etc/xdg/lxsession/LXDE-pi/autostart /home/pi/.config/lxsession/LXDE-pi/autostart

add at en of the file 

	sudo nano /home/pi/.config/lxsession/LXDE-pi/autostart

        /usr/bin/python3 /home/pi/pipresents/pipresents.py -o /home/pi -p pp_liveshow_1p4

save and exit

autorise pi user to execute pipresents at startup

	sudo apt-get install xserver-xorg-legacy
	sudo nano /etc/X11/Xwrapper.config

modify line : 
  allowed_users = anybody

Set a static ip
If you are using ethernet

	 ip route | grep eth0

If you are using wifi 

	 ip route | grep wlan0

	 sudo nano /etc/dhcpcd.conf

Add lines (get ip from overhead response)

	interface wlan0
	static ip_address=xxx.xxx.xxx.xxx/24
	static routers=xxx.xxx.xxx.xxx.

Restart raspberry

connect ssh to PIBOOTH Rasberry to generate a private/public key without passphrase

	ssh-keygen -t rsa

_Generating public/private rsa key pair._
_Enter file in which to save the key (/home/pi/.ssh/id_rsa):_

 Press return

_Created directory '/home/pi/.ssh'._
_Enter passphrase (empty for no passphrase):_

 Press return

_Enter same passphrase again:_

 Press return

_Your identification has been saved in /home/pi/.ssh/id_rsa......_

Copy your identification key to the raspberry pipresents (modify ip xxx.xxx.xxx.xxx to pipresents ip)

	ssh-copy-id -i ~/.ssh/id_rsa.pub pi@xxx.xxx.xxx.xxx

You will have to confirm (yes) and login with the password of the pibooth "pi" user
now you can login ssh from pibooth to pipresents without password (ssh pi@xxx.xxx.xxx.xxx)

Rsync the pibooth folder from Pibooth to the /home/pi/pipresents/pp_home/pp_live_tracks (only .jpg files, but not file with 0bytes because it crashes the pibooth display)

	rsync -e ssh -o -avz --min-size=1 /home/pi/Pictures/pibooth/*.jpg pi@xxx.xxx.xxx.xxx:/home/pi/pp_home/pp_live_tracks

Create a script file copy2pipresents.sh

	sudo nano copy2pipresents.sh

add these lines (the # at next line has to be copied in the file (shabang)

```
  #!/bin/bash
  rsync -e ssh -o -avz --min-size=1 /home/pi/Pictures/pibooth/*.jpg pi@10.10.0.3:/home/pi/pp_home/pp_live_tracks &
```

save and exit
change owner and rights

	sudo chown root:root copy2pipresents.sh
	sudo chmod 777 copy2pipresents.sh

to rsync after each picture add a plugin in 

	 sudo nano /usr/local/lib/python3.7/dist-packages/pibooth/plugins/pibooth_copy2pipresents.py

Paste the code :
```

"""Plugin to rsync pictures to pipresents"""
import os
import pibooth
from pibooth.utils import LOGGER

__version__ ="0.0.1"

@pibooth.hookimpl
def state_processing_exit(app):
    name = os.path.basename(app.previous_picture_file)
    
    #copy file to pipresents
    LOGGER.info('transfert du fichier vers l écran distant')
    os.popen('/home/pi/copy2pipresents.sh')
```

save and exit

activate the plugin in the config file of pibooth or in the general plugin menu
Your pictures should be send to pipresents each time pibooth saves a picture and the slideshow restarts backwards automatically

I added a cron on pipresents raspberry to erase the pictures at each startup, so you do not have to empty the pp_live_tracks each time you use the pipresents.

	crontab -e

add 

  @reboot rm /home/pi/pp_home/pp_live_tracks/*.jpg

anxuae, 2023-05-04:
Hi @dominiquegrota 

Thanks a lot! I will add this toto to the official documentation.

anxuae, 2023-05-11:
Hi @dominiquegrota 

I'm write the step-by-step procedure in the documentation: https://pibooth.readthedocs.io/en/test-camera/sources/install/slideshow.html
I've made some changes. Feel free to comment.

dominiquegrota, 2023-05-11:
Hi @anxuae, 
I just looked at the documentation, and it is clear ! Much better than I originaly wrote it ! Thank you for the job ! 
I hope some users will have fun with it ! 
You can cose this issue.

closed 2023-05-12 by anxuae

srvoets, 2024-06-12:
> Hi @dominiquegrota
> 
> I'm write the step-by-step procedure in the documentation: https://pibooth.readthedocs.io/en/test-camera/sources/install/slideshow.html I've made some changes. Feel free to comment.

These doc links are 404ing and I see no mention of 'slideshow' in the docs.

dominiquegrota, 2024-07-12:
@anxuae, could you have a look at the docs you added one year ago, maybe this has not been merged to the project, or loss in an other branch ? 
Thank's
Dominique

werdeil, 2024-07-13:
This documentation is on a stale branch, the corresponding documentation was no longer available, I relaunched its build, the link it working again.

P.S. You may see that it redirect you to https://documentation.pibooth.org/en/test-camera/sources/install/slideshow.html as documentation is now embedded in the pibooth.org website.