pibooth/pibooth #472: Pibooth crash, possible FTP error, but not sure.

MikkeBoomBoom, 2023-05-10:
Thanks a lot for a great piece of software, I have been playing with photobooths for years, but this is simply a different ballgame, so happy to have found pibooth :) 

Running Picam v2 on Rpi 4.
Plugins: Ftp, Google Photo (Deactivated) and QR.
everything working fine most of the time, but sometimes I get the following error at the last stage, and it's occuring more often lately:
 
 [ INFO    ] pibooth           : Creating the final picture
[ INFO    ] pibooth           : Use OpenCvPictureFactory to create background
[ INFO    ] pibooth           : Use OpenCvPictureFactory to concatenate images
[ INFO    ] pibooth           : Use OpenCvPictureFactory to assemble final image
[ INFO    ] pibooth           : Use OpenCvPictureFactory to draw texts
[ INFO    ] pibooth           : Save image '/home/pi/Pictures/pibooth/2023-05-10-20-27-32_pibooth.jpg'
[ ERROR   ] pibooth           : 421 No transfer timeout (300 seconds): closing control connection
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/states.py", line 76, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_callers.py", line 60, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "/home/pi/Desktop/photoboothplugins/pibooth_ftp.py", line 23, in state_processing_exit
    app.ftp.storbinary('STOR {}'.format(name), fp, 1024)
  File "/usr/lib/python3.7/ftplib.py", line 503, in storbinary
    self.voidcmd('TYPE I')
  File "/usr/lib/python3.7/ftplib.py", line 278, in voidcmd
    return self.voidresp()
  File "/usr/lib/python3.7/ftplib.py", line 251, in voidresp
    resp = self.getresp()
  File "/usr/lib/python3.7/ftplib.py", line 244, in getresp
    raise error_temp(resp)
ftplib.error_temp: 421 No transfer timeout (300 seconds): closing control connection
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='pi', release='5.10.103-v7l+', version='#1529 SMP Tue Mar 8 12:24:00 GMT 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.174 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'

anxuae, 2023-05-11:
This is clearly an issue for your FTP server. It means that the connection falls in timeout.
Have a look on this post https://stackoverflow.com/questions/33269971/python-error-handling-for-ftp-hang-421-timeout

Probably, you could find a solution on the net

Regards

MikkeBoomBoom, 2023-05-11:
Alright, thanks for the reply, and thanks again for an awesome piece of software.