pibooth/pibooth #474: pibooth does not work with DSLR camera

oli1204, 2023-05-22:
Hello,

i installed Pibooth on my Raspberry. With the Raspberry Pi-Cam it works without problems. But with my Canon EOS 350D camera it will not work.
It takes a picture, but it seems it can not download it.
When i take a photo with Gphoto2 the camera works, i can take a picture and can download it.

I make a test and this is the result:
*===============================================================================
Testing commands used by pibooth
Setting option imgsettings/iso="100"
   -> unsupported setting imgsettings/iso=100 (nothing configured on DSLR)
Setting option settings/capturetarget="Memory card"
   -> unsupported setting settings/capturetarget=Memory card (nothing configured on DSLR)
Unknown option actions/viewfinder
Take capture preview
Take a capture
Download file from DSLR
Gphoto2: b'gp_context_error': b'canon_usb_set_file_attributes: canon_usb_dialogue failed'
Save capture locally from memory buffer

================================================================================
SUCCESS : diagnostic completed
If you are investigating why pibooth does not work with your DSLR camera,
please paste the content of generated file 'diagnostic.log'
on https://github.com/pibooth/pibooth/issues

sorry i don't do that many issues on github, i hope i created this correctly and someone is here who can help me

oli1204, 2023-05-22:
My diagnostic logfile

`Pibooth version installed: 2.0.7
Installed plugins: picture-template-1.1.0
GPhoto2 version installed: 2.5.27
  - standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
  - gcc (C compiler used)
  - ltdl (for portable loading of camlibs)
  - EXIF (for special handling of EXIF files)
Listing all connected DSLR camera
00 : addr-> usb:001,009  name-> Canon Digital Rebel XT (normal mode)

================================================================================
Stating diagnostic of connected DSLR camera
* Preview compatible: True
* Capture compatible: True
/settings/ownername
  Label       : Owner Name
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Text
  Current     : Viktor Joos
  Choices     : n/a
/settings/capturesizeclass
  Label       : Capture Size Class
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Menu widget (same as Radio)
  Current     : Compatibility Mode
  Choices     : ['Compatibility Mode', 'Thumbnail', 'Full Image']
/settings/iso
  Label       : ISO Speed
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Menu widget (same as Radio)
  Current     : 160
  Choices     : ['50', '100', '125', '160', '200', '250', '320', '400', '500', '640', '800', '1000', '1250', '1600', '3200']
/settings/shootingmode
  Label       : Shooting mode
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Menu widget (same as Radio)
  Current     : A-DEP
  Choices     : ['AUTO', 'P', 'Tv', 'Av', 'M', 'A-DEP', 'M-DEP', 'Bulb', 'Manual 2', 'Far scene', 'Fast shutter', 'Slow shutter', 'Night scene', 'Gray scale', 'Sepia', 'Portrait', 'Spot', 'Macro', 'BW', 'PanFocus', 'Vivid', 'Neutral', 'Flash off', 'Long shutter', 'Super macro', 'Foliage', 'Indoor', 'Fireworks', 'Beach', 'Underwater', 'Snow', 'Kids and pets', 'Night snapshot', 'Digital macro', 'MyColors', 'Photo in movie']
/settings/shutterspeed
  Label       : Shutter Speed
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Menu widget (same as Radio)
  Current     : 1/200
  Choices     : ['Bulb', '30', '25', '20', '15', '13', '10', '8', '6', '5', '4', '3.2', '2.5', '2', '1.6', '1.3', '1', '0.8', '0.6', '0.5', '0.4', '0.3', '1/4', '1/5', '1/6', '1/8', '1/10', '1/13', '1/15', '1/20', '1/25', '1/30', '1/40', '1/50', '1/60', '1/80', '1/100', '1/125', '1/160', '1/200', '1/250', '1/320', '1/400', '1/500', '1/640', '1/800', '1/1000', '1/1250', '1/1600', '1/2000', '1/2500', '1/3200', '1/4000', '1/5000', '1/6400', '1/8000']
/settings/aperture
  Label       : Aperture
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Menu widget (same as Radio)
  Current     : 20
  Choices     : ['1.2', '1.4', '1.6', '1.8', '2.0', '2.2', '2.5', '2.8', '3.2', '3.5', '4.0', '4.5', '5.0', '5.6', '6.3', '7.1', '8', '9', '10', '11', '13', '14', '16', '18', '20', '22', '25', '29', '32']
/settings/exposurecompensation
  Label       : Exposure Compensation
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Menu widget (same as Radio)
  Current     : -2/3
  Choices     : ['+2', '+1 2/3', '+1 1/2', '+1 1/3', '+1', '+2/3', '+1/2', '+1/3', '0', '-1/3', '-1/2', '-2/3', '-1', '-1 1/3', '-1 1/2', '-1 2/3', '-2']
/settings/imageformat
  Label       : Image Format
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Menu widget (same as Radio)
[... 60 lines cut ...]
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 0
  Choices     : [0, 1]

================================================================================
Testing commands used by pibooth
Setting option imgsettings/iso="100"
   -> unsupported setting imgsettings/iso=100 (nothing configured on DSLR)
Setting option settings/capturetarget="Memory card"
   -> unsupported setting settings/capturetarget=Memory card (nothing configured on DSLR)
Unknown option actions/viewfinder
Take capture preview
Take a capture
Download file from DSLR
Gphoto2: b'gp_context_error': b'canon_usb_set_file_attributes: canon_usb_dialogue failed'
Save capture locally from memory buffer

================================================================================
SUCCESS : diagnostic completed
If you are investigating why pibooth does not work with your DSLR camera,
please paste the content of generated file 'diagnostic.log'
on https://github.com/pibooth/pibooth/issues
`

anxuae, 2023-06-28:
Hi @oli1204 

Does the `pibooth-diag` command generate an picture ?

there is a GPhoto error in the log, so maybe the raison:

```
Gphoto2: b'gp_context_error': b'canon_usb_set_file_attributes: canon_usb_dialogue failed'
```