pibooth/pibooth #475: Does somebody knows what this plugin exactly does pibooth-qr-download

rubkec, 2023-05-23:
quesition, got everything working 
but installed also the following plugin
pibooth-qr-download

but is it mandatory? what does the plugin do?
how can it be uninstalled?

MikkeBoomBoom, 2023-05-23:
It provides a QR code on screen allowing the users to download the last picture taken. In my build I used it with the Dropbox integration, worked like a charm. 

If installed with pip you can uninstall like so: https://pip.pypa.io/en/stable/cli/pip_uninstall/ or you can disable the plugin in the config file or the in-app menu.

rubkec, 2023-05-23:
thnx but thats the pibooth-qr plugin, the plugin that i mean is https://pypi.org/project/pibooth-qr-download/0.1.2/#files

anxuae, 2023-06-03:
No idea, the best it to download the code and look inside

closed 2023-06-03 by anxuae