pibooth/pibooth #476: How to configure pibooth to print borderless with Canon Selphy CP1300

BeeRii, 2023-05-25:
Hello, i installed a Canon CP1500 and used the driver from the CP1300.

An image is printed successfully.

When I print out the second photo, the terminal shows:

Printer: Canon Selphy CP1500 processing

Only when I restart the printer i can I print the next image and the message Print job .... completed appears in the terminal.

Any ideas? Also checked issue #340 and #268

BeeRii, 2023-05-25:
It takes about 60-120 seconds for the status to go idle. If I wait this time, there is no need to restart the printer.

anxuae, 2023-06-03:
It seem an issue with the printer, or interaction with CUPS server.
I let the issue opened a time, if someone has an idea to help you

ceeeeb, 2023-09-13:
> Hello, i installed a Canon CP1500 and used the driver from the CP1300.
>
> An image is printed successfully.
>
> When I print out the second photo, the terminal shows:
>
> Printer: Canon Selphy CP1500 [...]

Hello BeeRii

Can you send me the procedure to configure the CP1500 with the CP1300 drivers?

Thanks

spideydigitalmaker, 2024-01-22:
Hello, 
I have the same configuration as @BeeRii and I've applied the modifications in the parameters, but nothing works, 
I can't print a photo until I reboot it.
Have you found a solution?
Thx

spideydigitalmaker, 2024-02-15:
I still have my problem. Can you help me?

VincentAulnay, 2024-03-04:
Same problem with my CP1500.
I have a border so big !! I need to delete it.
I tried every thing see here, anything works :(

Someone have found a solution.

poupounetjoyeux, 2024-04-15:
Hello all,

I just bought the CP 1500 to build a photobooth using pibooth for my weeding and had the same issue with bog borders but found a solution!

Firstly, I installed gutenprint from source to have the latest 5.3.4 version that is now supporting officially the CP 1500 for CUPS (you can find some tutorials on the web to do this)
When it was updated, I added m'y printer in CUPS using the CP1500 + guten print 5.3.4 driver.
This driver seems to be working only on USB and the printer display a data error if you try to print over WiFi but by going in admistration panel of CUPS and then selecting modify printer, a new driver appear in the list (CP1500 + guntenprint 5.3.4 IPP everywhere allowing less options but working with wifi and borderless format). I dont really know why it cannot be selected at setup time and only show the default CUPS ipp driver that is not respecting borders..

Once it is modified with the appropriate driver, go to the default options page of the printer and select the postcard.borderless format (check also in other tabs if there is a margin option to disable it, I dont remember and have not my PC next to me)

Now you can try without specifying any additional option from thé pibooth config (I have just put the printer name in it, that's all!)

For me it's now working without any borders and over Wifi, also tried with the pibooth template plugin and all is working!

Hope it can help 😉

spideydigitalmaker, 2024-04-18:
Thank you for your reply, I had managed to solve the problem the way you did. But I had forgotten to post the answer. 
Now I have to solve the problem of the lack of paper in the printer. When the printer runs out of paper, it is disabled in cups. You then have to restart it in CUPS, which is not possible during an event.

poupounetjoyeux, 2024-04-18:
> Thank you for your reply, I had managed to solve the problem the way you did. But I had forgotten to post the answer. Now I have to solve the problem of the lack of paper in the printer. When the [...]
Thank for your feedback,
I just tried on my side putting only one paper in the printer and cannot reproduce your issue..
If there is no paper, then the printer ask me to insert paper, if I insert it the job continue and next photos are correctly print
If I cancel the job, printer go back to idle state in CUPs and next photos also print correctly
Maybe there is a special scenario when not adding paper until a timeout is reached?

I'm on the latest raspbian so maybe this was fixed in latest packaged CUPs?
I also updated thé selphy printer to the latest firmware..

BeeRii, 2025-04-26:
> Hello all,
> 
> I just bought the CP 1500 to build a photobooth using pibooth for my weeding and had the same issue with big borders but found a solution!
> 
> Firstly, I installed gutenprint from source to have the latest **5.3.4** version that is now supporting officially the CP 1500 for CUPS (you can find some tutorials on the web to do this. I used the snapshot **gutenprint-5.3.4-2023-12-14T01-00-6a3da773.tar.xz** from sourceforge) When it was updated, I added m'y printer in CUPS using the **CP1500 + guten print 5.3.4** driver. This driver seems to be working only on USB and the printer display a data error if you try to print over WiFi but by going in admistration panel of CUPS and then selecting modify printer, a new driver appear in the list : **CP1500 + guntenprint 5.3.4 _IPP everywhere_** allowing less options but working with WiFi and borderless format. I dont really know why it cannot be selected at setup time and only show the default CUPS ipp driver that is not respecting borders..
> 
> Once it is modified with the appropriate driver, go to the default options page of the printer and select the **postcard.borderless** format (check also in other tabs if there is a margin option to disable it, I don't remember and have not my PC next to me)
> 
> Now you can try without specifying any additional option from the pibooth config (I have just put the printer name in it, that's all!)
> 
> For me it's now working without any borders and over WiFi. I also tried with the pibooth template plugin and all is working great!
> 
> Hope it can help 😉

BeeRii, 2025-04-26:
After a lot of trial and error, I found a solution and adjusted the driver slightly. I connect the printer and the Raspberry Pi to an access point. I still have to test the direct connection. Even when the paper is empty, the printer remains in idle mode and prints as soon as the paper is refilled.