pibooth/pibooth #477: pibooth myconfig1/   not working with installed plugins

DJ-Dingo, 2023-05-26:
Usually when I have used "pibooth myconfig1/" I have not had any plugins installed, but only had links to the plugins I used, in the cfg. So it's never been a problem in the past when I've done it that way.

Since my test machine (which runs bullseye 64-bit) and the plugins are already installed, and since the camera does not work in 64bit, I just used "pibooth_stillimage_camera". and it has been working fine until I wanted to make more configs.

If I use, for example, "pibooth myconfig1/", it says that my camera is not found. And then if I refer to "pibooth_stillimage_camera.py" in the myconfig1.cfg,,  it says "plugins already rigistered."

Am i doing something wrong here ?

**Environment**
 - OS bullseye 64
 - Raspberry Pi Camera Module v2.1
 - Pibooth version 2.7

anxuae, 2023-06-03:
Hi @DJ-Dingo 

How did you install "pibooth_stillimage_camera.py" plugin?
I don't know this plugin, but making several config should not have an impact.

anxuae, 2023-06-03:
It seems tat you have installed it with pip, as consequence you can not declare it again in the configuration. It's normal.

DJ-Dingo, 2023-06-04:
Okey thx. I did not know that.

closed 2023-06-04 by DJ-Dingo