pibooth/pibooth #480: Pibooth takes pictures from itself

Refurb, 2023-06-03:
**Describe the bug**
I have pibooth up and running and very happy with everything. However, sometimes it starts capturing from itself. I don't get where it's coming from. I'm using a Canon DSLR and a touchscreen without any physical buttons and lights. There isn't really a pattern in the behavior (for example let's say it's taking a picture from itself every 12 minutes). However, it seems that it only happens after you take a photo manually by yourself. If you just start the system and leave it without taking a picture manually, the automatically random capturing doesn't happen.

Of course, this is not what I want. Sometimes there are three ghost pictures in an hour, sometimes just 1 the entire evening. It's a waste of the paper.

**Logs**
I had to wait for it to occur again. This happened at the 3rd of June at 16.10 (4.30PM). You can check my logs here:
The log of pibooth (I did not copy the entire log because there were lots of lines, but you can start from the bottom and go up: https://pastebin.com/du3cZKhM
The config file: https://pastebin.com/2hbHqX5J
The diagnostic.log made: https://pastebin.com/H16UEDz6

I buy a beer for the one who fixes my problem! Let me know, many thanks!

anxuae, 2023-06-03:
Hi @Refurb 

Unfortunately your logs doesn't contains the moment when the capture is triggered. I see only the state 'wait', but no capture after that.

Is it only the DSLR that take a picture by itself? or the an entire pi booth sequence is triggered ()?

Refurb, 2023-06-03:
Hi @anxuae, thanks for your reply! Oh, do I need to provide another log?
When you search for "2023-06-03-16-10-37" you see several moments of the picture which was taken automatically.
It's the entire pibooth indeed, from capturing till printing.

anxuae, 2023-06-03:
Looking for "2023-06-03-16-10-37", I see the capture but not the event that triggered the sequence. I need several lines before your log

Refurb, 2023-06-04:
It was a big log to upload. You can download the entire log here: https://www.dropbox.com/s/zcjl94vh61o670c/pibooth.log?dl=0

Let me know, thanks!

Refurb, 2023-06-08:
Is this log better? Does someone has any idea? :-)

anxuae, 2023-06-28:
I @Refurb 

I can not see any trigger in the logs. I will try with my own setup without any physical button

anxuae, 2023-07-11:
Hi @TJay0 

It may be due to the PIN of the button ahas no state 1 or 0 because they are not connected to something with a pull-up/pull-down resistor. 
Somme electromagnet disturbance may induce random trigger, but is is juste a supposition.

TJay0, 2023-07-11:
Hi all,
I try to set the debounce_delay to a higher value and check the behavioural of the pibooth. Is the debounce_delay also applicable for the touchscreen input?

Regards,
TJ

anxuae, 2023-07-11:
No, debounce is only for hardware button

TJay0, 2023-07-20:
Hi all,

I changed the debounce delay to 0.5 and 0.8 but I can not see any difference in the random trigger event. The random trigger happen more or less some times in a hour. However, if I unplug the USB connection of the touchscreen it looks okay. So the touch input is coming from the display. I am not the software expert but it is possible to create a debounce for the touchscreen input?

Regards,
TJ

Refurb, 2023-08-23:
I've bought a new touch screen and it isn't happening anymore, I guess it was some kind of hardware bug.