pibooth/pibooth #481: Canon EOS 650D takes 4 shots instead of one

Grynr38, 2023-06-05:
Hi,
I have attached my 650D DSLR to a RPi 4, and for each and every photo it takes 4 shots (saving only one...) but it keeps taking 4 shots. So  when we click on "4 shots"... well it the shutter takes 4x4 ... 16 ! shots.
That "problem" is only found with pibooth. No problem whastoever with windows. I've tried PhotoboothProject.github.io too, same Pi, same DSRL, works as it should : just one shutter/shot per photo.
Any help would be greatly appreciated.

Thanks a lot.

Grynr38, 2023-06-05:
My bad... It seems to be caused by the preview...
Is there a way to only have 1 shutter preview, normal count-down and picture taken ? instead of 4 shutter previews ?
Or maybe a live preview ? or no preview at all ?

Thanks.

werdeil, 2023-06-05:
It is indeed linked to the preview. We didn't find a way to get video through Gphoto2 so we had to implement this fallback mechanism. My best advice is to use pibooth in hybrid mode: a Pi camera for the preview and the DSLR for the photo.

supercairos, 2023-07-19:
I've stumbled into the same issue and it was caused by OpenCV apt pkg using an old gphoto2 library (rather than the one build locally)

I would recommand uninstalling OpenCV using apt and installing it manually using pip

anxuae, 2023-07-20:
I @supercairos 

Thanks for the feedback.

During a long time, it was not possible to install OpenCV with pip on PiOS. Is now possible?
I could modify the setup to install it automatically using pip.

supercairos, 2023-07-20:
I believe so, I've used [opencv-python](https://pypi.org/project/opencv-python/)

But I might be in a special setup as I'm using a Rpi 4 with the 64bits and Debin Bullseye

supercairos, 2023-07-20:
Actually looking at the precompiled files, it seams there is no 32bits version for OpenCV python binding:
https://pypi.org/project/opencv-python/#files

adonno, 2024-04-08:
i've been trying with both openCV via apt and pip on Rpi 4 64bits Bullseye 

the one installed with apt allow me to start pibooth, but then i get an oups something went wrong. 
the one installed with pip doesn't start because i cannot find my canon 650D 

Is there a way to get the image of a working version ?  or explain the steps you did @supercairos ? that would be awesome

supercairos, 2024-04-08:
Hi @adonno,

I actually don't remember the proper steps, but I do remember that I built gphoto from sources. But in order to load it properly into PiBooth, you need to properly remove the `apt` version before (it is sometime installed by default). 

Maybe if you have some error logs, I can help troubleshoot them...

supercairos, 2024-04-08:
In order to install GPhoto, I've used:

```
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/.env
chmod +x gphoto2-updater.sh
sudo ./gphoto2-updater.sh
```

adonno, 2024-04-08:
I'll leave tonight for 3 days but i'll come back with logs when i'm back thank you. 

Can't wait to get it finally working and to start adding hardware buttons.

adonno, 2024-04-08:
> In order to install GPhoto, I've used:
> 
> 
> 
> ```
> 
> wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh
> 
> wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/.env
> 
> chmod +x gphoto2-updater.sh
> 
> sudo ./gphoto2-updater.sh
> 
> ```

Yup did that it's also in their doc

supercairos, 2024-04-08:
Check that the Gphoto PiBooth is loading is really the built version and not a PKG version.

A gnome picture utility is also using Gphoto so it's usually installed by default on rpi4...

supercairos, 2024-04-08:
Please output the pibooth-diag output also :)

adonno, 2024-04-12:
hi i'm finally back from vacation 
Like i said I installed the PIP version and have tried with legacy on and off from the raspi-config 

If I try with a webcam it works and the Camera is recognised on the PI since it shows the SD card 

```
adonno@pibooth:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ WARN:0@0.888] global cap_v4l.cpp:997 open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ERROR:0@0.893] global obsensor_uvc_stream_channel.cpp:159 getStreamChannelGroup Camera index out of range
[ WARN:0@0.894] global cap_v4l.cpp:997 open VIDEOIO(V4L2:/dev/video1): can't open camera by index
[ERROR:0@0.898] global obsensor_uvc_stream_channel.cpp:159 getStreamChannelGroup Camera index out of range
[ WARN:0@0.899] global cap_v4l.cpp:997 open VIDEOIO(V4L2:/dev/video2): can't open camera by index
[ERROR:0@0.903] global obsensor_uvc_stream_channel.cpp:159 getStreamChannelGroup Camera index out of range
/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py:469: PluggyTeardownRaisedWarning: A plugin raised an exception during an old-style hookwrapper teardown.
Plugin: pibooth-core:camera, Hook: pibooth_setup_camera
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected
For more information see https://pluggy.readthedocs.io/en/stable/api_reference.html#pluggy.PluggyTeardownRaisedWarning
  outcome = Result.from_call(
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 131, in __init__
    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 501, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 119, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 473, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 99, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 61, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 470, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 155, in _multicall
    teardown[0].send(outcome)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 28, in pibooth_setup_camera
    cam = camera.find_camera()
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/__init__.py", line 52, in find_camera
    raise EnvironmentError("Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected")
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected

```

supercairos, 2024-04-12:
Can you print the version of Gphoto ?

`gphoto2 --version`

adonno, 2024-04-12:
```
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

```

```
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
Pibooth version installed: 2.0.8
Installed plugins: 
gPhoto2 not installed, cannot diagnose connected DSLR
```

hm this is wierd gphoto2 not installed ?

adonno, 2024-04-12:
btw thank you for your quick response

supercairos, 2024-04-12:
Yes, there might be an error here. 

You're on `Bookworm` `64bits` right ?

supercairos, 2024-04-12:
You can see here the version of the software It's building from source:
https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/.env

supercairos, 2024-04-12:
Good luck 🙏

adonno, 2024-04-12:
I worked 👍  now i'm tempted to try on a pi5 😄

supercairos, 2024-04-12:
Glad it worked! 
On my camera (700D) putting it on video mode gave a much better frame-rate than in picture mode.

I haven't tried on a pi5 (yet) but I didn't felt it was the RPi 4 that was limiting the camera fps.

Would love some feedback on a pi5 tho :)

adonno, 2024-04-12:
It worked as a cahrm on a pi 5

now that i have both working i will test the speed difference 
regarding the frame-rate, i'm using a webcam for previews and it works like. charm actually, much faster pictures since it doesn't need to open /close the mirror for previews. Only point now is to find a webcam that has the almost same zoom level as the canon lens i'm using