pibooth/pibooth #482: Raspi Cam V3

TJay0, 2023-06-06:
Hello together,

I ordered the new RaspiCam V3 for my pibooth. I tried to replace the existing Cam V2.1 one by one with the Cam V3 with a full update. However, it didn't work :). I can take a preview with the libcamera-hello with the V3 and I get a preview video. So the camera V3 works on the RPI. In pibooth I get the error "Can not get camera preview image" when i try to start a countdown for the preview. Is the RaspiCam V3 already supported or do I need a workaround for the V3?  

Regards,
TJ

jrperk, 2023-06-07:
I have the same issue.

agrovista, 2023-06-16:
new V3 camera needs Bullseye version of the os to run
current version of pi booth is not supported on bullseye, buster does not contain the libraries to run the v3 camera 
version 3 of pibooth is said to have support for the new camera modules

dmlogic, 2023-06-16:
> version 3 of pibooth is said to have support for the new camera modules
That's helpful, thank you

pirlgon, 2023-06-16:
With the new V3 cam and the version 3 of pibooth i get a preview is not started error

pirlgon, 2023-06-16:
I have a RPI 4 1GB and the v3 cam

Here is what i get when i try to take a picture:
```
$ pibooth -v
<frozen importlib._bootstrap>:228: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected
pygame 2.4.0 (SDL 2.0.14, Python 3.9.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.3.9
[ INFO    ] pibooth           : Installed plugins:
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
libpng warning: iCCP: CRC error
libpng warning: iCCP: CRC error
[ DEBUG   ] pibooth           : Use pibooth camera management system
[ DEBUG   ] pibooth           : Try to kill process 'gvfs-gphoto2-volume-monitor'
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Failed to allocate required memory.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[0:11:13.073030073] [1990]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:11:13.203991117] [2003]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media1
[ DEBUG   ] pibooth           : Found libcamera cameras:
[ DEBUG   ] pibooth           :   > 00 : name-> imx708_wide | addr-> 2
[0:11:13.208696814] [1990]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:11:13.346346366] [2006]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media1
[ INFO    ] picamera2.picamera2: Initialization successful.
[ INFO    ] picamera2.picamera2: Camera now open.
[ DEBUG   ] picamera2.picamera2: <libcamera._libcamera.CameraManager object at 0xcb95b6c0>
[ INFO    ] pibooth           : Configuring Libcamera camera ...
[ DEBUG   ] picamera2.picamera2: Camera was not started
[ DEBUG   ] picamera2.picamera2: Requesting configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'display': 'main', 'encode': 'main'}
[0:11:13.364148084] [1990]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-XBGR8888
[0:11:13.364546390] [2006]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
[ INFO    ] picamera2.picamera2: Configuration successful!
[ DEBUG   ] picamera2.picamera2: Final configuration: {'use_case': 'preview', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (640, 480), 'stride': 2560, 'framesize': 1228800}, 'lores': None, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Minimal: 3>, 'FrameDurationLimits': (100, 83333)}, 'display': 'main', 'encode': 'main'}
[ DEBUG   ] picamera2.picamera2: Streams: {'main': <libcamera._libcamera.Stream object at 0xcb95be00>, 'lores': None, 'raw': None}
[ DEBUG   ] picamera2.picamera2: Allocated 4 buffers for stream 0.
[ INFO    ] picamera2.picamera2: Camera started
[ DEBUG   ] pibooth           : Use pibooth printer management system
[ INFO    ] pibooth           : Connected to printer 'HP_DeskJet_3700_series_E9298D_'
[ DEBUG   ] pibooth           : +<- Activate state 'wait'
[ DEBUG   ] pibooth           : |   Set scene 'wait'
libpng warning: iCCP: CRC error
libpng warning: iCCP: CRC error
[ DEBUG   ] pibooth           : +-> took 1.778 seconds
[ DEBUG   ] pibooth           : +<- Activate state 'choose'
[ DEBUG   ] pibooth           : |   Set scene 'choose'
[ INFO    ] pibooth           : |   Show picture choice (nothing selected)
[ DEBUG   ] pibooth           : +-> took 1.813 seconds
[ DEBUG   ] pibooth           : +<- Activate state 'chosen'
[ DEBUG   ] pibooth           : |   Set scene 'chosen'
[ DEBUG   ] pibooth           : |   No translation defined for 'en/0' key
[ INFO    ] pibooth           : |   Show picture choice (1 captures selected)
[ DEBUG   ] pibooth           : +-> took 4.024 seconds
[ DEBUG   ] pibooth           : +<- Activate state 'preview'
[ DEBUG   ] pibooth           : |   Set scene 'preview'
[ INFO    ] pibooth           : |   Show preview before next capture
[ ERROR   ] pibooth           : |   Preview is not started
[ DEBUG   ] pibooth           : |   Back to failsafe state due to error:
Traceback (most recent call last):
  File "/home/pirlgon/photobooth/pibooth/pibooth/states.py", line 58, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 80, in _hookexec
[... cut ...]
[0:07:03.826635195] [1801]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:07:03.950917273] [1803]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media1
[0:07:03.951820332] [1801]  INFO Camera camera.cpp:1028 configuring streams: (0) 2304x1296-YUV420
[0:07:03.952160159] [1803]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
```

werdeil, 2023-06-19:
Hi all, as stated by @agrovista raspi cam v3 needs libcamera to run and this is the main reason pibooth is not compatible with Bullseye.
We indeed plan to release a version 3 of pibooth including compatibility with Bullseye/libcamera/raspiV3 but it is still ongoing. It is mainly done by @anxuae and it is not yet finished. We hope to have a beta version soon.

pirlgon, 2023-06-20:
Hi @werdeil and thanks for your reply, I was testing the 3.x branch, if we can help in any way it would be a pleasure.

jrperk, 2023-06-20:
I have all three Pi camera versions plus a V3 wide and Pi3, 4, and 400s and 7-inch touchscreens.  If you need to test any particular combination, I would be happy to help.

anxuae, 2023-06-28:
Pi camera  V3 and Pi Os Bullseyeare not yes supported, see #290

closed 2023-06-28 by anxuae