pibooth/pibooth #484: Print error window pop up

sy6sy2, 2023-06-10:
**Describe the bug**
I use a Canon CP1300 connected over USB.
Everything works great until there is no more paper or ink.
In this case, Pibooth move into background and I go back into the Pi desktop with a Print error pop up window (see picture).
Is somebody knows how to disable this window pop up?

Also, once I add more paper I need to manually un-pause the printer on CUPS side (like https://github.com/pibooth/pibooth/issues/296) but it's another story and I can apply some trick to fix.
The main issue at the moment is the Printer error message coming in foreground.-.

Thanks

**Steps to reproduce the bug**
1. Use a Canon Selphy CP1300
2. Use Pibooth and print photos until there is no more paper in the printer cassette
3. A "Printer error" window pop up appears and Pibooth fullscreen window move in background

**Environment**
 - OS: Raspbian Buster
 - Camera: Rpi camera
 - Pibooth version: 2.0.7

sy6sy2, 2023-06-11:
Seems like removing `system-config-printer` package did the trick.

werdeil, 2023-06-19:
Thanks, do you know if this package is installed by default? We never had this issue before.

sy6sy2, 2023-06-19:
I guess because I never install it and I run pibooth on a fresh PiOS install

Momonom123, 2023-11-03:
I have the same problem, but I didn't get what I have to do to fix it. Could you please discripe your solution a bit more detailed? it would help me a lot

sy6sy2, 2023-11-03:
You can try a `sudo apt remove system-config-printer`

Momonom123, 2023-11-03:
Thank you for the quick response. The window does not pop up anymore, but the printer is still paused and has to be manually un-pause in CUPS when adding new paper

sy6sy2, 2023-11-03:
Yes, I have the exactly same problem with a Canon CP1300.
Let me explain, this is the issue I had:

* The printer works as expected...
* Then, at a moment, the printer goes out of paper (or out of ink)
* In the background (since we do not have the pop window anymore) CUPS pause the printer
* Then I add more paper in the printer, and the printer detect new paper and print the "blocked" last picture
* BUT, it means that printer driver is broken (or something like that) and the printer does not tell CUPS that it is ready, so CUPS leaves the printer in pause and any new taken photo is not printed anymore even if the printer has paper...
* To fix this, you have to manually un-paused the printer in CUPS, as you said...

This is a big issue with or usage in PiBooth because we do not want the user to open the CUPS webpage to un pause the printer manually.

After a lot of research I did not find any solution to fix this bug.

So, the first and "hacky" solution is to run a script in background that will periodically check if the printer is paused, un unpause it.

The second solution (this is what I did) is to add a dedicated physical button on your photobooth and bind this button press to run a bash or python script that will unpause the printer.
So, in practice, once I added mode paper or replace the ink I only need to press this button and my Photobooth is ready again.

You will find below the scripts that I use, fill free to adapt for you:
(Also, in my script, when I unpause the printer, I remove the latest job to prevent duplicate print after the paper recharge).

```python3
#!/usr/bin/env python3

#  Copy the content of this file in /home/pi/scripts/listen_for_unpause_printer.py

# Add this line in /etc/rc.local:
# /home/pi/scripts/listen_for_unpause_printer.py &

import logging
import subprocess
import sys
import time
from logging.handlers import SysLogHandler

from gpiozero import Button
from unpause_printer import unpause_printer

button = Button(8)

logging.basicConfig(
    level=logging.INFO,
    handlers=[
        SysLogHandler(
            facility=SysLogHandler.LOG_DAEMON,
            address='/dev/log'
        ),
        logging.StreamHandler(stream=sys.stdout)
    ],
    format="%(asctime)s - %(filename)s:%(funcName)s:%(lineno)d %(levelname)s - '%(message)s'"
)

def main():
    logging.info("Starting...")
    while True:
        try:
            logging.info("Waiting for press...")
            button.wait_for_press()
            logging.info("Button pressed")
            unpause_printer()
            time.sleep(0.5)
        except KeyboardInterrupt:
            logging.info("KeyboardInterrupt")
            break
        except Exception as e:
            logging.info("Exception: " + str(e))
            break

if __name__ == '__main__':
    logging.info("Starting...")
    main()
    logging.info("Exiting...")
```

```python3
#!/usr/bin/env python3

# Copy the content of this file in  /home/pi/scripts/unpause_printer.py

# Add this line in /etc/rc.local (so that we try to unpause the printer on PiBooth start up):
# /home/pi/scripts/unpause_printer.py &

import logging
import subprocess
import sys
import time
from logging.handlers import SysLogHandler

logging.basicConfig(
    level=logging.INFO,
    handlers=[
        SysLogHandler(
            facility=SysLogHandler.LOG_DAEMON,
            address='/dev/log'
        ),
        logging.StreamHandler(stream=sys.stdout)
    ],
    format="%(asctime)s - %(filename)s:%(funcName)s:%(lineno)d %(levelname)s - '%(message)s'"
)

def run_cmd(cmd):
    try:
        result = subprocess.run(cmd, stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
        try:
            return result.stdout.decode('utf-8')
        except Exception as e:
            return result.stdout
    except Exception as e:
        logging.error("Failed to run command {}: {}".format(" ".join(cmd), str(e)))
    return ""

def is_cp1300_disabled() -> bool:
    try:
        logging.info("Getting CP1300 printer status...")
        stdout = run_cmd(["/usr/bin/lpstat", "-p"])
        for line in stdout.splitlines():
            if "cp1300" in line.lower():
                if "disabled since" in line.lower():
                    logging.info("CP1300 printer is disabled")
                    return True
                logging.info("CP1300 printer is enabled")
                return False

        logging.error("CP1300 printer not found")
        return False

    except Exception as e:
        logging.error("Failed to get CP1300 printer status: " + str(e))
        return False

def enable_cp1300():
    try:
        logging.info("Re-enabling printer...")
        r = run_cmd(["/usr/sbin/cupsenable", "Canon_SELPHY_CP1300"])
        logging.info('Re-enable result: ' + r)
    except Exception as e:
        logging.error("Failed to enable CP1300: " + str(e))

def remove_oldest_job():
    try:
        jobs = run_cmd(["/usr/bin/lpstat"])

        if jobs == "":
            logging.info("No pending jobs")
            return

        logging.info("Pending jobs:")
        logging.info(jobs)

        oldest_job = jobs.partition('\n')[0].split(" ")[0]
        logging.info("Oldest job: " + oldest_job)

        logging.info("Canceling oldest job...")
        r = run_cmd(["/usr/bin/cancel", oldest_job])
        logging.info('Cancel job result: ' + r)
    except Exception as e:
        logging.error("Failed to remove oldest job: " + str(e))

def unpause_printer():
    cp1300_disabled = is_cp1300_disabled()

    if cp1300_disabled:
        logging.info("We need to remove oldest job (if not yet done) and re-enable printer")
        remove_oldest_job()

        enable_cp1300()

def main():
    unpause_printer()

if __name__ == '__main__':
    logging.info("Starting...")
    time.sleep(5)
    main()
    logging.info("Exiting...")

```

Momonom123, 2023-11-05:
thank you so much. I tryed to solve the problem in a simular way. I wrote a plugin that trys to unpause the printer every time a picture was taken and is processed. I have to test now if this solution is sufficient or if it leads to other problems.
Here is what I did:

"""Plugin to unpause the printer when entering in the 'processing' state."""

import pibooth
from pibooth.utils import LOGGER
import logging
import os

__version__ = "1.0.0"

@pibooth.hookimpl
def state_processing_enter():
    try:
        os.system('/usr/sbin/cupsenable Canon_CP910')
    except Exception as e:
        logging.error("Failed to enable CP910: " + str(e))

Momonom123, 2023-11-06:
it seems that this solution works well, but not perfectly. My solution has the problem that I don't delete the last job (because I honestly don't know how to do that). Therefore, the image that was taken when the paper or ink was empty is printed twice.

sy6sy2, 2023-11-06:
I had the same problem as you: just after a paper recharge, the printer start printing the « blocked » picture. But (I don’t know what) this same picture is still in the jobs queue, so when you unpause the printer on CUPS the printer will print again the same picture….

This is why in my script I remove the last job just before unpausing the printer, this way I do not have this « printed twice » issue. 

Check the « remove_oldest_job » function in my script to adapt in your case. 

I guess that in your case you should adapt your hook this way:
« If printer is paused, remove oldest job and unpause the printer »

djimkanters, 2024-07-16:
Having the same issue. The sudo remove system-config-printer did not work for me. Anyone found a better way? 
[image]

djimkanters, 2024-07-18:
Combined your ideas in the following code

`import logging
import subprocess
import sys
import time
from logging.handlers import SysLogHandler
import pibooth
from pibooth.utils import LOGGER

# Logging setup
logging.basicConfig(
    level=logging.INFO,
    handlers=[
        SysLogHandler(
            facility=SysLogHandler.LOG_DAEMON,
            address='/dev/log'
        ),
        logging.StreamHandler(stream=sys.stdout)
    ],
    format="%(asctime)s - %(filename)s:%(funcName)s:%(lineno)d %(levelname)s - '%(message)s'"
)

PRINTER_NAME = "Canon_SELPHY_CP1300"

def run_cmd(cmd):
    try:
        result = subprocess.run(cmd, stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
        try:
            return result.stdout.decode('utf-8')
        except Exception as e:
            return result.stdout
    except Exception as e:
        logging.error("Failed to run command {}: {}".format(" ".join(cmd), str(e)))
    return ""

def is_printer_disabled() -> bool:
    try:
        logging.info("Getting printer status...")
        stdout = run_cmd(["/usr/bin/lpstat", "-p", PRINTER_NAME])
        for line in stdout.splitlines():
            if PRINTER_NAME.lower() in line.lower():
                if "disabled since" in line.lower():
                    logging.info(f"{PRINTER_NAME} is disabled")
                    return True
                logging.info(f"{PRINTER_NAME} is enabled")
                return False
        logging.error(f"{PRINTER_NAME} not found")
        return False
    except Exception as e:
        logging.error("Failed to get printer status: " + str(e))
        return False

def enable_printer():
    try:
        logging.info("Re-enabling printer...")
        r = run_cmd(["/usr/sbin/cupsenable", PRINTER_NAME])
        logging.info('Re-enable result: ' + r)
    except Exception as e:
        logging.error(f"Failed to enable {PRINTER_NAME}: " + str(e))

def remove_oldest_job():
    try:
        jobs = run_cmd(["/usr/bin/lpstat"])

        if not jobs:
            logging.info("No pending jobs")
            return

        logging.info("Pending jobs:")
        logging.info(jobs)

        oldest_job = jobs.partition('\n')[0].split(" ")[0]
        logging.info("Oldest job: " + oldest_job)

        logging.info("Canceling oldest job...")
        r = run_cmd(["/usr/bin/cancel", oldest_job])
        logging.info('Cancel job result: ' + r)
    except Exception as e:
        logging.error("Failed to remove oldest job: " + str(e))

def unpause_printer():
    if is_printer_disabled():
        logging.info("We need to remove oldest job (if not yet done) and re-enable printer")
        remove_oldest_job()
        enable_printer()

@pibooth.hookimpl
def state_processing_enter():
    try:
        unpause_printer()
    except Exception as e:
        logging.error("Error in state_processing_enter: " + str(e))

if __name__ == '__main__':
    logging.info("Starting...")
    time.sleep(5)
    unpause_printer()
    logging.info("Exiting...")`

Also edited openbox to never switch focus.