pibooth/pibooth #487: problem with pibooth crash

evabgrt, 2023-06-17:
Hello, I have a big problem, please save me !!!!!

pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (961) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1214) setProperty OpenCV | GStreamer warning: GStreamer: unhandled property
[ WARNING ] pibooth           : No printer configured in CUPS (see http://localhost:631)
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (536) startPipeline OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1085) setProperty OpenCV | GStreamer warning: no pipeline
[ ERROR   ] pibooth           : Can not capture frame
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='6.1.21-v7+', version='#1642 SMP Mon Apr  3 17:20:52 BST 2023', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created

werdeil, 2023-06-19:
Hi @evabgrt , can you confirm you are not running on Bullseye OS? Pibooth is not compatible with this OS.

evabgrt, 2023-06-24:
> Salut@evabgrt, pouvez-vous confirmer que vous n'utilisez pas Bullseye OS ? Pibooth n'est pas compatible avec ce système d'exploitation.

thanks u !! that is my problem, it's okay now !

closed 2023-06-24 by evabgrt