pibooth/pibooth #488: GIF and Boomerang effect

Lescoal, 2023-06-19:
Hello there, hello @anxuae, hello @werdeil 

Following your answer here : https://github.com/pibooth/pibooth/issues/42#issuecomment-1402633427, a feature proposal would be a boomerang effect and a gif effect in addition to captures ranging from 1 to 4 photos. I already have the python code for both features that is working but I don't know how to integrate them yet.
Here is the "gif" code that is working perfectly in standalone : 
`import time
import os
from datetime import datetime
from picamera import PiCamera
from PIL import Image

# Initialize the camera
camera = PiCamera()

# Start the preview
camera.start_preview()

# Wait for the user to adjust the camera
input("Press enter to start taking photos...")

# Take 3 photos with a 2 second delay between each photo
for i in range(3):
    camera.capture('photo{}.jpg'.format(i))
    time.sleep(2)

# Stop the preview
camera.stop_preview()

# Create a GIF from the 3 photos
with open('animation.gif', 'wb') as f:
    images = [Image.open(f'photo{i}.jpg') for i in range(3)]
    images[0].save(f, save_all=True, append_images=images[1:], duration=400, loop=0)

# Get the current date and time
now = datetime.now()

# Format the date and time as a string
timestamp = now.strftime("%Y-%m-%d_%H-%M-%S")

# Define the path where you want to save the gif
path = '/home/pi/Pictures/gif/'

# Save the GIF locally with the timestamp as the file name in the specific path
with open(path + f'animation_{timestamp}.gif', 'wb') as f:
    images = [Image.open(f'photo{i}.jpg') for i in range(3)]
    images[0].save(f, save_all=True, append_images=images[1:], duration=400, loop=0)

# Delete the original animation.gif
os.remove('animation.gif')

`

And here is boomerang code that is pretty slow in term of general duration process time so I'm not sure it can be implemented yet, probably it need a little improvment : 
`import datetime
import subprocess
from picamera import PiCamera
import os

# Initialize the camera
camera = PiCamera(resolution=(1920, 1080))

# Start the preview
camera.start_preview()

# Wait for the user to adjust the camera
input("Press enter to start recording...")

# Start recording a 5-second video
camera.start_recording('video.h264')
camera.wait_recording(3)
camera.stop_recording()

# Stop the preview
camera.stop_preview()

# Get the current date and time
now = datetime.datetime.now()

# Format the date and time as a string
timestamp = now.strftime("%Y-%m-%d_%H-%M-%S")

# Define the path where you want to save the boomerang video
path = '/home/pi/Pictures/boomerang/'

# Use FFmpeg to reverse the video and create a reversed copy of the video
subprocess.call(['ffmpeg', '-i', 'video.h264', '-vf', 'reverse', '-c:a', 'copy', 'reversed_video.h264'])

# Concatenate the reversed video with the original video to create a boomerang effect
subprocess.call(['ffmpeg', '-i', 'concat:video.h264|reversed_video.h264', '-c:v', 'h264', '-c:a', 'aac', '-filter:v', 'setpts=0.5*PTS', path + 'boomerang_' + timestamp + '.mp4'])

# Delete the original video and reversed video
os.remove('video.h264')
os.remove('reversed_video.h264')

`

Amazing project and amazing work you did there as I already said !

Regards

anxuae, 2023-06-28:
Hi @Lescoal 

Thanks a lot for your code, we could integrate it in a plugin or directly in pibooth, what do you think @werdeil ?

So to be sure, you don't want to display it at screen, just generate gif or video in the saved directory ?
(displaying it at screen will be more tricky)

marwiede, 2023-06-28:
Hi @anxuae, hi @Lescoal, 

while looking for another topic, I found your discussion here. I came across the gif feature at another photoboot last weekend and thought about integrating it. I have one early version which is realized as Plugin. The plugin checks if a series of images was captured and creates a gif then. Is this relevant for you or is another implementation more useful? If this matches the requested feature and you like some contribution, I can continue the work next weekend and share it here.

Btw. @anxuae pibooth is an amazing project, I really enjoy building my own photobooth with it! Thanks!

werdeil, 2023-07-04:
Hi all I'm just wondering if we shall not merge this mechanism (generate a gif of x photos) to the animate photo mechanism already in place (that displays the 4 photos as a gif to the screen but keep the generated image with the 4 frames). In this case @marwiede mechanism could be done directly (if there are a series of image it generates a gif out of it).

For the Boomerang part, it is a little bit different as we are taking a video not a group of photos. Maybe a plugin for this yes.

marwiede, 2023-07-05:
Yeah I was not aware of this animate mechanism. Sounds reasonable! 
Nevertheless, was a good experience for me doing my first custom plugin.