pibooth/pibooth #491: Is it possible to remove option to choose 1 or 4 photos?

stre1026, 2023-06-30:
Hello! Thank you for this great piece of software. 

Is it possible to remove a state? I'd like to not give the option of choosing between 1 and 4 photos and go right into taking 4 photos. I looked at the code and I'm having a hard time figuring out how it all works together. Part of the issue is I'm not a Python developer. I have taken many programming classes over the years, I'm just not a master of any of them :)

werdeil, 2023-06-30:
Hi @stre1026 thanks for the question.

It is possible to force the choice, you just need to put it as the only option in the config file :

Change the line
```
captures = (4, 1)
```
To 
```
captures = (4)
```
And it will skip the "choose" screen and go directly to the capture steps.

stre1026, 2023-06-30:
Oh, that's perfect. Thank you!!

werdeil, 2023-07-03:
Closing it as it seems to be answered

closed 2023-07-03 by werdeil