pibooth/pibooth #493: Newly setup pibooth crashes with PIL.Image ANTIALIAS error

macprince, 2023-07-02:
**Describe the bug**
I've just set up pibooth according to the instructions on https://pibooth.readthedocs.io/en/stable/sources/install.html#install but when I try to start it, it crashes with a "module 'PIL.Image' has no attribute 'ANTIALIAS'" error. 

**Logs**
[pibooth.log](https://github.com/pibooth/pibooth/files/11931390/pibooth.log)

**Environment**
 - OS: Raspbian Bullseye on a RPi 4B 8GB
 - Camera: RPi HQ Camera v1.0
 - Pibooth version: 2.0.7

werdeil, 2023-07-03:
Hi @macprince pibooth is not yet compatible with Raspios Bullseye and its libcamera, could you try to reinstall on Buster? Thanks

macprince, 2023-07-04:
Hi, @werdeil. Yep, that was it, I wiped that card back to Buster and it works fine there.

closed 2023-07-04 by macprince