pibooth/pibooth #495: module 'PIL.Image' has no attribute 'ANTIALIAS'

joby29, 2023-07-05:
HI
i trie to install pibooth but i have theses errors when i launch it ...

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 98, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 433, in __call__
    return self._hookexec(self.name, self._hookimpls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 112, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 418, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 108, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 70, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 415, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 116, in _multicall
    raise exception.with_traceback(exception.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 80, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/view_plugin.py", line 53, in state_wait_enter
    win.show_intro(previous_picture, app.printer.is_ready()
  File "/usr/local/lib/python3.9/dist-packages/pibooth/view/window.py", line 249, in show_intro
    self._update_background(background.IntroBackground(self.arrow_location, self.arrow_offset))
  File "/usr/local/lib/python3.9/dist-packages/pibooth/view/window.py", line 125, in _update_background
    self._current_background.resize(self.surface)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/view/background.py", line 219, in resize
    self.left_arrow = pictures.get_pygame_image("arrow.png", size, vflip=vflip, color=self._text_color)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/pictures/__init__.py", line 97, in get_pygame_image
    Image.ANTIALIAS if antialiasing else Image.NEAREST)
AttributeError: module 'PIL.Image' has no attribute 'ANTIALIAS'
[ DEBUG   ] pibooth           : took 0.023 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='6.1.36-v7+', version='#1659 SMP Fri Jun 30 13:13:14 BST 2023', machine='armv7l'>
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *

thanks for your help.

anxuae, 2023-07-05:
Hi @joby29 

Can you tell me what is the version of PIL library?

striggy85, 2023-07-08:
Hi!

Same problem here. 

My version of PIL library is 10.0.0

anxuae, 2023-07-11:
PIL 10.0.0 has introduced some incompatibilities.
We need to fix it.
You can downgrade PIL version to 9 to make it working

jacare0000, 2023-07-16:
HI!

The same happen to me. 
As anxuae said version 9 solve the isue.
only run 
"pip install Pillow==9.5.0"

anxuae, 2023-07-20:
Fixed with v2.0.8

closed 2023-07-20 by anxuae