pibooth/pibooth #496: Pibooth is starting random capture events

TJay0, 2023-07-11:
Hello together,

I have an interresting behavioral with my pibooth. Sometimes the pibooth is doing some random capture events without any input from the touchscreen. Have someone the same "ghost events" to trigger a picture? It looks like a print event will be triggered as well, so i have limited the printer to 1 duplicate. I have saved the config and the log file in the attachment, maybe one of you find an error :)

**Environment**
 - Camera Rpi camera V2 + Nikon D3200
 - Waveshare 13inch Touchscreen
 - Citizen CZ-01
 - Pibooth version 2.0.6

[diagnostic.log](https://github.com/pibooth/pibooth/files/12018011/diagnostic.log)
[pibooth.log](https://github.com/pibooth/pibooth/files/12018012/pibooth.log)

anxuae, 2023-07-11:
Hi @TJay0 

Thanks for reporting, this issue is a duplicate of #480, we have not found any reason for this bug.
Do you have physical button connected to your pibooth?

closed 2023-07-11 by anxuae

TJay0, 2023-07-11:
Hi @anxuae 

I have only a touchscreen used for an input. Sometimes it triggers 6 times in a row and sometimes 1 in two hours. Or is a touchscreen debouncing necessary?

Regards,
TJ

anxuae, 2023-07-11:
No hardware button is the common point with #480
So I guess issue is link to that, can we continue the discussion in the other issue?