pibooth/pibooth #498: Problem green capture

evabgrt, 2023-07-12:
Hello, i'm using Facecam elgato camera. I have a problem with it, I can take pictures but the layout is green, what can I do, please? 

It is not the webcam, it's fonctional with "cheese"

Thank you

anxuae, 2023-07-20:
Hi @evabgrt 

Can you provide log?

What is the version of pibooth? Version of PIL library?

evabgrt, 2023-07-31:
Hello, sorry for the time of my response, i was in vacations...

pibooth
/usr/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected. Consider enabling compile time detection with environment variables like PYGAME_DETECT_AVX2=1 if you are compiling without cross compilation.
  return f(*args, **kwds)
pygame 2.5.0 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins:
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ INFO    ] pibooth           : Configuring OpenCV camera ...
VIDEOIO ERROR: V4L2: setting property #30 is not supported
[ INFO    ] pibooth           : Connected to printer 'EPSON_WF_3520_Series'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ INFO    ] pibooth           : Saving raw captures
[ INFO    ] pibooth           : Creating the final picture
[ INFO    ] pibooth           : Use OpenCvPictureFactory to create background
[ INFO    ] pibooth           : Use OpenCvPictureFactory to concatenate images
[ INFO    ] pibooth           : Use OpenCvPictureFactory to assemble final image
[ INFO    ] pibooth           : Use OpenCvPictureFactory to draw texts
[ INFO    ] pibooth           : Save image '/home/pi/Pictures/pibooth/2023-07-31-14-36-14_pibooth.jpg'
[ INFO    ] pibooth           : Display the final picture

version Pibooth 2.0.8
version Pillow 9.2.0

evabgrt, 2023-07-31:
> Hi @evabgrt
> 
> Can you provide log?
> Hello, sorry for the time of my response, i was in vacations...

pibooth
/usr/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected. Consider enabling compile time detection with environment variables like PYGAME_DETECT_AVX2=1 if you are compiling without cross compilation.
return f(*args, **kwds)
pygame 2.5.0 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO ] pibooth : Installed plugins:
[ INFO ] pibooth : Starting the photo booth application on Raspberry pi 4B
[ INFO ] pibooth : Configuring OpenCV camera ...
VIDEOIO ERROR: V4L2: setting property https://github.com/pibooth/pibooth/issues/30 is not supported
[ INFO ] pibooth : Connected to printer 'EPSON_WF_3520_Series'
[ INFO ] pibooth : Show picture choice (nothing selected)
[ INFO ] pibooth : Show picture choice (1 captures selected)
[ INFO ] pibooth : Show preview before next capture
[ INFO ] pibooth : Take a capture
[ INFO ] pibooth : Saving raw captures
[ INFO ] pibooth : Creating the final picture
[ INFO ] pibooth : Use OpenCvPictureFactory to create background
[ INFO ] pibooth : Use OpenCvPictureFactory to concatenate images
[ INFO ] pibooth : Use OpenCvPictureFactory to assemble final image
[ INFO ] pibooth : Use OpenCvPictureFactory to draw texts
[ INFO ] pibooth : Save image '/home/pi/Pictures/pibooth/2023-07-31-14-36-14_pibooth.jpg'
[ INFO ] pibooth : Display the final picture

version Pibooth 2.0.8
version Pillow 9.2.0
> What is the version of pibooth? Version of PIL library?