pibooth/pibooth #499: Pibooth crashes before going to the picture preview screen

MatthiasPG, 2023-07-12:
Pibooth does not go to the preview screen, instead it says that something went wrong and goes back to the start screen. Devices: Raspberry Pi 4 and Cannon EOS 500D.
gphoto2 capture and download works

Diagnosis log:

Pibooth version installed: 2.0.7
Installed plugins: 
GPhoto2 version installed: 2.5.27
  - standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
  - gcc (C compiler used)
  - ltdl (for portable loading of camlibs)
  - EXIF (for special handling of EXIF files)
Listing all connected DSLR camera
00 : addr-> usb:001,007  name-> Canon EOS 500D

================================================================================
Stating diagnostic of connected DSLR camera
* Preview compatible: True
* Capture compatible: True
/actions/syncdatetime
  Label       : Synchronize camera date and time with PC
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 0
  Choices     : [0, 1]
/actions/bulb
  Label       : Bulb Mode
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 2
  Choices     : [0, 1]
/actions/uilock
  Label       : UI Lock
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 2
  Choices     : [0, 1]
/actions/autofocusdrive
  Label       : Drive Canon DSLR Autofocus
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 0
  Choices     : [0, 1]
/actions/manualfocusdrive
  Label       : Drive Canon DSLR Manual focus
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Radio button
  Current     : None
  Choices     : ['Near 1', 'Near 2', 'Near 3', 'None', 'Far 1', 'Far 2', 'Far 3']
/actions/cancelautofocus
  Label       : Cancel Canon DSLR Autofocus
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 0
  Choices     : [0, 1]
/actions/eoszoom
  Label       : Canon EOS Zoom
  Readonly    : no
  Data type   : <class 'str'>
  Widget type : Text
  Current     : 0
  Choices     : n/a
/actions/eoszoomposition
  Label       : Canon EOS Zoom Position
[... 416 lines cut ...]
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'camera_capture_preview [library.c:3326]': b"'ret' failed: get_viewfinder_image failed (0xa102: Camera Not Ready)"
Gphoto2: b'gp_camera_capture_preview [gphoto2-camera.c:1403]': b"'camera->functions->capture_preview ( camera, file, context)' failed: -1"

================================================================================
ABORT   : exception occures: [-1] Unspecified error
If you are investigating why pibooth does not work with your DSLR camera,
please paste the content of generated file 'diagnostic.log'
on https://github.com/pibooth/pibooth/issues

anxuae, 2023-07-20:
Hi @MatthiasPG 

I can see in the log that your auto focus is enabled, which cause probably too long response time from DSLR.

Please set your DSLR in manual mode.

```
/capturesettings/autoexposuremode
Label : Canon Auto Exposure Mode
Readonly : no
Data type : <class 'str'>
Widget type : Radio button
Current : Green
Choices : ['P', 'TV', 'AV', 'Manual', 'Bulb', 'A_DEP', 'DEP', 'Custom', 'Lock', 'Green', 'Night Portrait', 'Sports', 'Portrait', 'Landscape', 'Closeup', 'Flash Off', 'Auto', 'Handheld Night Scene', 'HDR Backlight Control', 'Food', 'Grainy B/W', 'Soft focus', 'Toy camera effect', 'Fish-eye effect', 'Water painting effect', 'Miniature effect', 'HDR art standard', 'HDR art vivid', 'HDR art bold', 'HDR art embossed', 'Panning', 'Self Portrait', 'Hybrid Auto', 'Smooth skin', 'Fv']
```

bigbeka, 2023-08-08:
@anxuae thank you for an amazing software! But it is proving to be difficult to get nice photos with auto focus disabled. Many photos are being taken out of focus as the Aperture is small, and depth of field is short.

many solution/workaround that could be implemented?
Perhaps tell the next step to wait before taking the picture while the camera is focusing? Maybe a config to set how long to wait until taking a photo?

Crapout, 2023-10-13:
Hello,

I'm facing the same issue, which is random, depending on the autofocus needed time (depending the current brightness, the flash trigger...).

After digging into gphoto2 issues, the answer i have found in many threads is to add a delay right before the capture parameter. The result command looks like : gphoto2 --wait-event=500ms --capture-image-and-download

@anxuae Could it be possible to add a "before_capture_delay" config into the .cfg file ?
- 0 if unused
- float value accepted

This feature will bring a huge value to this amazing tool you guys made, as picture quality will be much improved 🙏 

Thank you 👍

tonarec, 2023-12-18:
Hi! I have the Canon EOS 2000D on a Raspberry PI 3B, and with a proper installation I'm able to take multiples photos with auto focus enabled. Make sure that gvfs is disable before connecting your camera with:
```
systemctl --user stop gvfs-gphoto2-volume-monitor.service
systemctl --user disable gvfs-gphoto2-volume-monitor.service
```
I'm running the Raspberry Pi OS (Legacy) Buster from 2023-05-03 (since other distro doesn't works flawlessly)
Here is my current installation:
```
Linux raspberrypi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```
Hope it helps