pibooth/pibooth #502: Help please. Pi4 4GB

mtz1734, 2023-07-24:
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ DEBUG   ] pibooth           : Preview resolution is (1920.0, 1080.0)
[ INFO    ] pibooth           : Connected to printer 'PDF'
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : took 13.938 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : took 4.085 seconds
[ DEBUG   ] pibooth           : Activate state 'preview'
[ INFO    ] pibooth           : Show preview before next capture
[ DEBUG   ] pibooth           : Add to buffer the image '3542801784'
[ DEBUG   ] pibooth           : No translation defined for 'de/capture' key
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 5653
[ DEBUG   ] pibooth           : Add to buffer the image '3542627784'
[ DEBUG   ] pibooth           : Add to buffer the image '3542802384'
[ DEBUG   ] pibooth           : Add to buffer the image '3542627808'
[ DEBUG   ] pibooth           : Add to buffer the image '3542801784'
[ DEBUG   ] pibooth           : Add to buffer the image '3542627784'
[ DEBUG   ] pibooth           : Add to buffer the image '3542802384'
[ DEBUG   ] pibooth           : Add to buffer the image '3542627688'
[ DEBUG   ] pibooth           : Add to buffer the image '3542801784'
[ DEBUG   ] pibooth           : Add to buffer the image '3542627808'
[ DEBUG   ] pibooth           : Add to buffer the image '3542802384'
[ DEBUG   ] pibooth           : Add to buffer the image '3542627688'
[ DEBUG   ] pibooth           : Add to buffer the image '3542802384'
[ DEBUG   ] pibooth           : Add to buffer the image '3542628048'
[ DEBUG   ] pibooth           : Add to buffer the image '3542801784'
[ DEBUG   ] pibooth           : Add to buffer the image '3542627688'
[ DEBUG   ] pibooth           : Add to buffer the image '3542801784'
[ DEBUG   ] pibooth           : took 3.400 seconds
[ DEBUG   ] pibooth           : Activate state 'capture'
[ INFO    ] pibooth           : Take a capture
[ DEBUG   ] pibooth           : Taking capture at resolution (1934, 2464)
[ ERROR   ] pibooth           : Can not capture frame
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 53, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 433, in __call__
    return self._hookexec(self.name, self._hookimpls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 112, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 418, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 108, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 70, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 415, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 116, in _multicall
    raise exception.with_traceback(exception.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 80, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 103, in state_capture_do
    app.camera.capture(effect)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 220, in capture
    raise IOError("Can not capture frame")
OSError: Can not capture frame
[ DEBUG   ] pibooth           : took 0.592 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='pi4test', release='6.1.21-v8+', version='#1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023', machine='aarch64', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.024 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'

supercairos, 2023-08-03:
Please provide the output of `pibooth-diag` cmd

anxuae, 2023-09-15:
Hello,

`pibooth` not compatible with PiOS Bullseye (downgrade to Buster)

See #290

closed 2023-09-15 by anxuae