pibooth/pibooth #504: Waiting Time after Printing

mtz1734, 2023-08-09:
How can I make a Plugin (or manipulate the source Code) to add a specific waiting Time (with loading Screen) after Printing?

What is the state for that?

Thanks so much
Mathias

dominiquegrota, 2023-08-09:
Hi, 
There is no specific status after printing (I think it could be added in next release ?). I wrote a Plugin to add a message on screen "printing in progress". This is not a official Plugin, and maybe it works not in all specific configuration. 
But if you want I can share it, and you can test.
Dominique

mtz1734, 2023-08-09:
That would bei great. Thank you so much!

How can you share it with me? GitHub? Or do you want a Cloud Link for Upload?

Thank you!

dominiquegrota, 2023-08-09:
I can share it here, it may be intresting for others ;-)
I have the code at home... il can't share it before tonight...
Dominique

dominiquegrota, 2023-08-09:
To add a new plugin named "pibooth_message_impression"
`sudo nano /usr/local/lib/python3.7/dist-packages/pibooth/plugins/pibooth_message_impression.py`

here is the plugin code : 
```
"""Plugin to display 'print in progress' after printing button fired"""

import time
import pygame
import pibooth
from pibooth import pictures, fonts
from pibooth.utils import LOGGER
import os

__version__ = "0.0.1"

@pibooth.hookimpl
def pibooth_startup(app):
    """init remaining_prints"""
    app.print_ok = 0

@pibooth.hookimpl
def state_processing_exit(app):
    """Actions performed when application exit Processing state"""
    app.print_ok = app.count.remaining_duplicates

@pibooth.hookimpl
def state_wait_do(win, app, events):
    #LOGGER.info(app.count.remaining_duplicates)
    #test after process state if remaining duplicates > 0 and if print event on print button is fired
    if app.find_print_event(events) != None and app.print_ok != 0 and app.find_print_event(events).printer == 1:
        LOGGER.info("sate_print_do en cours !")
		
        # remove 1 print form remaing duplicates
        app.print_ok = app.print_ok - 1
        LOGGER.info(app.print_ok)
		
        #show text printing in progress on screen
        win_rect = win.get_rect()
        text = "Printing in progress !!!"

        # Get best font size according to window size
        font = fonts.get_pygame_font(text, fonts.CURRENT,
                                     win_rect.width//3, win_rect.height//2)

        # Build a surface to display at screen
        text_surface = font.render(text, True, (255, 255, 255))

        # Draw the surface at screen
        # add the text at 1000px from left ant 880 px form top of screen
        win.surface.blit(text_surface, (1100, 880))

        # Force screen update and events process
        pygame.display.update()
        pygame.event.pump()

        # Wait 4s before delete message
        time.sleep(4)
```

mtz1734, 2023-08-13:
Another question:

How can I change the code so that instead of a time elapsing before the message disappears, a key has to be pressed? Or both hardware buttons?

dominiquegrota, 2023-08-30:
This plugin is a workaround, there is no status for the printing execution. I think it will be hard to wait for a key to be pressed. And I do not really understand why you need this ? The printing progress can continue while you take a new picture with the app, isn't it working for you ?

mtz1734, 2023-09-06:
This workaround works perfectly for me.

but I want an additional, different message for the paper refill process:
When the print counter reaches the refill level (after 19 prints), it should display a message with a "Done" button or something similar.

Like this:
https://github.com/pibooth/pibooth/issues/414
   Thanks

dominiquegrota, 2023-09-06:
ok, I understand...

I do not have my pibooth actually, but I guess you could merge the 2 plugins...
You may test this : 

`"""Plugin to display 'refill paper' for canon Selfy cp1300 'print in progress' after printing button fired"""

import time
import pygame
import pibooth
from pibooth import pictures, fonts
from pibooth.utils import LOGGER
import os

__version__ = "0.0.1"

text_error = "Plus de papier, ou plus d'encre !    Réapprovisionnez"
text_done = "Terminé"
path_image = "YOUR-PICTURE-NAME.png"

@pibooth.hookimpl
def pibooth_startup(app):
    """init remaining_prints"""
    app.print_ok = 0

@pibooth.hookimpl
def state_processing_exit(app):
    """Actions performed when application exit Processing state"""
    app.print_ok = app.count.remaining_duplicates

@pibooth.hookimpl
def state_wait_do(win, app, cfg,events):
    if app.count.printed >= cfg.getint('PRINTER', 'max_pages') and cfg.getint('PRINTER', 'max_pages') != -1:
        
        win_rect = win.surface.get_rect()

        # Get best font size according to window size
        font = fonts.get_pygame_font(text_error, fonts.CURRENT,
                                    win_rect.width//1.5, win_rect.height//1.5)

        # Clear screen
        if isinstance(win.bg_color, (tuple, list)):
            win.surface.fill(win.bg_color)
        else:
            bg_surface = pictures.get_pygame_image(win.bg_color, win_rect.size, crop=True, color=None)
            win.surface.blit(bg_surface, (0, 0))

        # Draw the surface at screen
        text_surface = font.render(text_error, True, win.text_color)
        pos = (win_rect.center[0]- win_rect.width//3, win_rect.top + win_rect.height//4)
        win.surface.blit(text_surface, pos)
        
        # Draw Done button
        text_surface = font.render(text_done, True, win.text_color)
        pos = (win_rect.center[0]+ win_rect.width*0.2,win_rect.center[1] + win_rect.height*0.15)
        win.surface.blit(text_surface, pos)
        
        # Draw picture
        size = [win_rect.width//2, win_rect.height//2]
        picture = pictures.get_pygame_image(path_image, size, hflip=False, vflip=False, color= win.text_color)
        x = int(win_rect.left)  # x = int(win_rect.left + win_rect.width * 0.70 - picture.get_rect().width//2)
        y = int(win_rect.bottom - size[1])  # y = int(win_rect.top + win_rect.height * 0.45)
        picture_pos = (x,y)
        win.surface.blit(picture, picture_pos)

        # Force screen update and events process
        pygame.display.update()
        pygame.event.pump()
        events = pygame.event.get()
        
        # wait for usre confirmation
        if app.find_capture_event(events) != None or app.find_print_event(events) != None:
            # reset printed images counter
            app.count.printed = 0

    #LOGGER.info(app.count.remaining_duplicates)
    #test after process state if remaining duplicates > 0 and if print event on print button is fired
    if app.find_print_event(events) != None and app.print_ok != 0 and app.find_print_event(events).printer == 1:
        LOGGER.info("sate_print_do en cours !")
		
        # remove 1 print form remaing duplicates
        app.print_ok = app.print_ok - 1
        LOGGER.info(app.print_ok)
		
        #show text printing in progress on screen
        win_rect = win.get_rect()
        text = "Printing in progress !!!"

        # Get best font size according to window size
        font = fonts.get_pygame_font(text, fonts.CURRENT,
                                     win_rect.width//3, win_rect.height//2)

        # Build a surface to display at screen
        text_surface = font.render(text, True, (255, 255, 255))

        # Draw the surface at screen
        # add the text at 1000px from left ant 880 px form top of screen
        win.surface.blit(text_surface, (1100, 880))

        # Force screen update and events process
        pygame.display.update()
        pygame.event.pump()

        # Wait 4s before delete message
        time.sleep(4)`