pibooth/pibooth #513: Stays stuck in Preview when Focus cant be found

osnabooth, 2023-08-30:
**Describe the bug**
Pibooth works great with PiCam as Previewcam and Nikon D90 as Main Camera. Its set on Autofocus and works great. When it cant find the focus the Preview screen "Smile" stays until i reboot. No errors and the Preview Image is still "live". 

**Steps to reproduce the bug**
1. Take a Picture
2. dont let him find focus (e.g. forget the Cap on the DSLR)

**Environment**
 - OS Raspbian Buster updated 30.08.23
 - Rpi camera + Nikon D90

werdeil, 2023-08-30:
Hi @osnabooth, the issue you get is linked to gphoto, that's the main reason we advise to disable autofocus.

We cannot do much on pibooth side, gphoto raises an error and pibooth catches it.

osnabooth, 2023-08-31:
> Hi @osnabooth, the issue you get is linked to gphoto, that's the main reason we advise to disable autofocus.
> 
> We cannot do much on pibooth side, gphoto raises an error and pibooth catches it.

Hi! Thanks for the fast answer. Isnt there a way to avoid being stuck in the preview by forcing "oops something went wrong" after x seconds of no photo taken? I understand that if autofocus isnt working the Problem appears. But I get way better Photos with AF since people tend to change Positions. They always have to restart the whole booth when the focus isnt working.

karosai, 2024-05-29:
Hi, 
I also have the same problem with my D90.
I think I found a solution I switch the Autofocus mode of my D90 on AF-C. It's priority to capture rather than focus. With this mode I no more have the "Oops something went wrong".
Best regards,
Simon