pibooth/pibooth #515: PiBooth + Pi Camera Module 3 Crashes on Picture Capture

Lu-cas-Rec, 2023-09-07:
**Describe the bug**
Hello, whenever I installed PiBooth I installed the tools in this order:
sudo apt-get install -y libsdl2-*
sudo apt-get install python3-opencv
pip install pibooth
Then I installed a Google Photos plugin:
pip3 install pibooth-google-photo

Whenever I launch PiBooth, I never see previews, and whenever I try to capture it crashes.
Thank you!

**Steps to reproduce the bug**
1. Launch PiBooth with "pibooth"
2. Try to take photo
3. See error.

**Logs**
1. Set the `debug` mode in the config - done
2. Post your config file - attached
3. Copy-paste all the logs from the console:
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: google-photo-1.2.3
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Failed to allocate required memory.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ INFO    ] pibooth           : Initialize Google Photos connection
[ DEBUG   ] urllib3.connectionpool: Starting new HTTPS connection (1): photos.google.com:443
[ DEBUG   ] urllib3.connectionpool: https://photos.google.com:443 "HEAD / HTTP/1.1" 302 0
[ DEBUG   ] urllib3.util.retry: Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
[ DEBUG   ] pibooth           : +< Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : +> took 10.678 seconds
[ DEBUG   ] pibooth           : +< Activate state 'choose'
[ INFO    ] pibooth           : |  Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : +> took 3.027 seconds
[ DEBUG   ] pibooth           : +< Activate state 'chosen'
[ INFO    ] pibooth           : |  Show picture choice (1 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : +> took 4.189 seconds
[ DEBUG   ] pibooth           : +< Activate state 'preview'
[ INFO    ] pibooth           : |  Show preview before next capture
[ ERROR   ] pibooth           : |  Can not get camera preview image
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 402, in main_loop
    self._machine.process(events)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 67, in process
    self.set_state(new_state_name)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 98, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 493, in __call__
    return self._hookexec(self.name, self._hookimpls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 115, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 457, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 114, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 76, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 454, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 113, in _multicall
    raise exception.with_traceback(exception.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 77, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 71, in state_preview_enter
    app.camera.preview(win)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 155, in preview
    self._window.show_image(self._get_preview_image())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 101, in _get_preview_image
    raise IOError("Can not get camera preview image")
OSError: Can not get camera preview image
[ ERROR   ] pibooth           : |  system='Linux', node='raspberrypi', release='6.1.21-v8+', version='#1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023', machine='aarch64', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

**Environment**
 - OS -  Raspbian Bullseye 64 Bit
 - Camera - Rpi camera Module 3
 - Pibooth version - 2.0.8

Lu-cas-Rec, 2023-09-07:
I just saw that you only support V2.1 so I purchased this Camera off of Amazon.
https://a.co/d/d6qsALn
Going to use this instead. Will you notify us through the Wiki whenever the Module 3 is supported?
Thanks

anxuae, 2023-09-15:
Hello,

`pibooth` not compatible with PiOS Bullseye nor PiCamera v3

See #290

closed 2023-09-15 by anxuae