pibooth/pibooth #518: Orientetion of Picture during Preview

Trunc4te, 2023-09-19:
**Describe the bug**
Give a clear and concise description of what the bug is.

During prieview the orientation of the camera picture is wrong. I switched the orientation in the OS, because i mounted the Monitor in Portrait (See attached Pictures)
After the prieview, everything is fine.

**Steps to reproduce the bug**
1. Set OS orientation top 'right'
2. Take a Picture with pibooth

**Environment**
 - OS: Debian
 - Camera: Raspberry Pi 12 MP HQ 
 - Pibooth version: 2.0.8

Trunc4te, 2023-09-28:
i think the problem is the Orientation of the OS

omgsideburns, 2023-12-30:
I'm having this same issue with a Picam v2.1.  With a usb webcam the preview was fine. I'm wondering if it has to do with GPU preview of the picamera..

omgsideburns, 2023-12-30:
Tested it with the webcam again, centered the preview just fine. Only gets quirky with the picam.

omgsideburns, 2023-12-30:
Tested with screen in normal orientation, works fine. Switching back to 90* screen rotation, the issue persists. 

I'm guessing it has to do with how pybooth is creating the rect that the preview displays on is centered.  Does the rpicam use a different process to do this than opencv cameras do?

omgsideburns, 2023-12-30:
OK I've figured out part of it at least.. if we can find out how to implement.. someone smart please chime in.

The preview_window() values are (xpos, ypos, xres, yres).  So (0, 0, 640, 480) would put a 640x480 preview in the top left corner of your screen.  If you rotate the display 90 degrees, the (0, 0, x,x) continues to reference default orientations top left, so in a 90* rotation, it puts the preview at the bottom left corner. 

It appears the x,y location is calculated by the get_rect() function in base.py.  Since the XY coordinates of preview_window always originate from the default rotations 0,0.. you would need to invert the XY position. 

I don't know if this helps the countdown overlay, since it's not rotating with the display, but I believe it would fix it's location on the screen.

omgsideburns, 2023-12-31:
Tested and confirmed, swapping "centerx" and "centery" in the code below from cameras/base.py fixes the positioning issue.  The countdown overlay remains rotated for the normal display orientation after this fix.  I also realized white border/background usually around the preview is not there, just the preview. 

From 'base.py'
   def get_rect(self, max_size=None):
        """Return a Rect object (as defined in pygame) for resizing preview and images
        in order to fit to the defined window.
        """
        rect = self._window.get_rect(absolute=True)
        size = (rect.width - 2 * self._border, rect.height - 2 * self._border)
        if max_size:
            size = (min(size[0], max_size[0]), min(size[1], max_size[1]))
        res = sizing.new_size_keep_aspect_ratio(self.resolution, size)
        return pygame.Rect(rect.centerx - res[0] // 2, rect.centery - res[1] // 2, res[0], res[1])