pibooth/pibooth #520: Option to replace GPIO buttons with USB buttons

the3rdpedal, 2023-09-22:
**Is your feature request related to a problem? Please describe.**
I'm having issues with the GPIO button configuration, and also don't have all the required hardware.  I do, however, have a DragonRise USB joystick module that has arcade buttons.  It appears to read like a USB keyboard input.

**Describe the solution you'd like**
I'd like to be able to easily configure USB buttons (on a keyboard, controller, etc) to behave like the GPIO buttons are designed for.  

My coding skills are sub-par, but I'm trying to figure out how to access the USB inputs in the pibooth python code and replace the GPIO code, but so far I'm stuck.  Can anyone assist?  Has anyone done this already?

nsty, 2023-09-26:
I had a similar problem because I wanted to use wireless (hardware) buttons and the keyboard keys for control change from view to view. Only with GPIO buttons a two button control is possible at the moment.

See #524 for my implementation. You should be able to adapt the keys to your needs depending on the keys that are triggered by your joystick.

srvoets, 2023-10-18:
> I had a similar problem because I wanted to use wireless (hardware) buttons and the keyboard keys for control change from view to view. Only with GPIO buttons a two button control is possible at the [...]

I just came here searching for something like this thank you.

Id like to have the usual large arcade button(s) in a 3d printed enclosure, so I dont need to run wires to the pi, I figured this should be able to be done with an arduino or a pico that can emulate kb input.  

Totally forgot the normal buttons were GPO