pibooth/pibooth #523: normal or wide-angle camera

ManuFr54, 2023-09-23:
Hello and Thanks for this great software.
Should you choose a standard camera or a wide-angle?
I'm between the 2 versions of this camera:
https://www.kubii.com/en/official-raspberry-pi-accessories/3878-camera-module-v3-raspberry-pi-3272496313699.html
The standard 75° version is €30
The 120° wide-angle version is €39

pirlgon, 2023-09-28:
First of all, PyBooth is not yet compatible with Pycamera v3, but that's not the answer you're looking for :)

dominiquegrota, 2023-09-28:
Hi ! 
I use a standard camera PIcam V2 Cam and it is (for my use "wide angle enough"). Here is a picture tacken about 1.5/1.8 m from the pibooth with this cam.

[image]
Dominique

ManuFr54, 2023-09-28:
Hi and Thank you pirlgon and Dominique,
in this case, I think I will take v2.

closed 2023-09-28 by ManuFr54