pibooth/pibooth #525: crash

Frontkiller, 2023-09-26:
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Failed to allocate required memory.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (4 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : Can not get camera preview image
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='6.1.21-v8+', version='#1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023', machine='aarch64', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

Frontkiller, 2023-09-27:
Using following:

Raspberry pi 4B
Pi cam module 3
Bullseye

agrovista, 2023-09-30:
Need Buster for the os
and the V3 cam is not supported on the current version of pibooth

Frontkiller, 2023-09-30:
> Need Buster for the os
> 
> and the V3 cam is not supported on the current version of pibooth
> 
> 

Thanks ! 

Do you know if they are working on a version that works with V3 cam ?

Or do you know another booth that works with V3 ?

agrovista, 2023-10-01:
_Do you know if they are working on a version that works with V3 cam ?_
yep 
_Or do you know another booth that works with V3 ?_
i do not know

Frontkiller, 2023-10-01:
> _Do you know if they are working on a version that works with V3 cam ?_
> 
> yep 
> 
> _Or do you know another booth that works with V3 ?_
> 
> i do not know 

Thanks. Do they already have a release date for the version that works with pi cam V3 ?

agrovista, 2023-10-01:
above my pay grade ;)

fatgeek, 2023-10-01:
> > _Do you know if they are working on a version that works with V3 cam ?_
> > yep
> > _Or do you know another booth that works with V3 ?_
> > i do not know
> 
> Thanks. Do they already have a release date for the version that works with pi cam V3 ?

According to the Milestone, two months ago. I hope they have the time to get it out before Halloween, when lots of people set up Pibooths.

werdeil, 2023-10-01:
Hi @Frontkiller , as kindly answered by @fatgeek and @agrovista pibooth is indeed not (yet) compatible with bullseye and by consequence the V3 camera, we are working on it but it is taking more time than expected and we are unfortunately not available enough to work on it. But we hope to make it work one day.

Frontkiller, 2023-10-01:
> Hi @Frontkiller , as kindly answered by @fatgeek and @agrovista pibooth is indeed not (yet) compatible with bullseye and by consequence the V3 camera, we are working on it but it is taking more time [...]

Many thanks for the answer, I do understand that you guys have to find time to work on a new version. Keep up the good work ! I love the Pibooth interface!

mgrl, 2023-10-02:
> > Need Buster for the os
> > and the V3 cam is not supported on the current version of pibooth
> 
> Thanks !
> 
> Do you know if they are working on a version that works with V3 cam ?
> 
> Or do you know another booth that works with V3 ?

You might want to check out my PhotoBooth app or the photoboothproject: https://mgrl.github.io/photobooth-docs/features/

werdeil, 2024-02-02:
Closing this issue as the picamera v3 support is followed by #526 and new OS support by #540.

closed 2024-02-02 by werdeil