pibooth/pibooth #527: Print button missing

BramWerbrouck, 2023-10-14:
After taking a picture, I see the result screen with the message "Thank You" but no print button available and it returns automatically to the homepage (to take another picture)

what did do do wrong? or which setting(s) do I need you check?

Crapout, 2023-10-23:
Hello @BramWerbrouck 

First of all, can you check that the CUPS library is installed ? If you didn't install it previously, please run : 
`sudo apt-get install cups libcups2-dev
`

Then, check if your printer is detected / listed here : http://localhost:631/printers/

If the list is empty it means your printer isn't self detected and it needs to be setted up via http://localhost:631/admin/ > Add new printer.
If your printer is not on the default list you can add these packages to add many more.
```
sudo apt-get install libusb-1.0
sudo apt-get install printer-driver-gutenprint
```

If this didn't solve your issue, can you run the following commands and post the output here ? :
`pibooth-diag` (diagnostic.log)
`pibooth-printcfg --json` (copy paste terminal output)

See you,

BramWerbrouck, 2024-07-01:
After reinstalling CUPS, everything works well :-)