pibooth/pibooth #534: Take a picture from a plugin

Exitras, 2023-11-07:
Hi, I created a plugin to connect pibooth via API to other systems and it works like a charm! I would now also love to trigger taking a picture directly from the plugin based on certain conditions such as time (for example take a picture every 30 minutes to prevent the flash from going into standby). 

It should be a single picture (not 4 pictures) and the process after taking the picture should be the default process (the system should ask for prints after the picture has been taken). 

I'm rather new to Python and my current attempt did not work (I know this code would not run at all, its mainly about the last two lines).

import pygame
import pibooth

def state_wait_do(app):
    if timer.is_timeout(): #or any other condition
          capture_event = pygame.event.Event(pygame.KEYDOWN, key=pygame.K_p)  # Assuming 'p' is the capture key
          pygame.event.post(capture_event)
          

It always says something like: 
"An error occured: event object has no attribute capture"

Did any of you manage to take a picture directly from a plugin? If yes, would you please share the code or correct mine?

Cheers, 
Exitras

Exitras, 2024-01-18:
Hi again, 
here is the line of code that works within a plugin:

pygame.event.post(pygame.event.Event(pygame.USEREVENT + 1, capture = 1, printer = 0, button= app.buttons.capture))

Gope it helps!

srvoets, 2024-06-12:
> Hi, I created a plugin to connect pibooth via API to other systems and it works like a charm! I would now also love to trigger taking a picture directly from the plugin based on certain conditions [...]

Have you posted this plugin anywhere?  I would like to make a button that when pressed can snap a photo from a security camera in Frigate/Home assistant, and then process it in pibooth and upload to google album.

Can I accomplish this with your plugin?