pibooth/pibooth #536: "Oops! after hours of working fine

myownhierarchy, 2023-11-18:
Hey folks,
after hours of working just fine, pibooth gets stuck in "get ready screen" and then shows the "Oops!" message. 
The first time this happened at a halloween party and I was able to reproduce this today when I was running pibooth for a few hours.
The log reads like it's actually a gphoto problem, but maybe one of you can point me in the right direction, so this won't happen every time. 
Thanks in advance!
[pibooth.log](https://github.com/pibooth/pibooth/files/13401016/pibooth.log)

werdeil, 2024-02-02:
Hi @myownhierarchy, sorry for the late reply. Did you managed to find a solution? 

From the logs, it is indeed linked to gphoto, maybe your dslr has an auto-sleep/auto-off feature that makes it unavailable after some time?

myownhierarchy, 2024-02-03:
Thank you for replying. Yes I indeed found out that my camera reset the settings for auto-sleep and I strongly think that was the issue.
You can consider this issue closed.

myownhierarchy, 2024-02-03:
Thank you for replying. Yes I indeed found out that my camera reset the settings for auto-sleep and I strongly think that was the issue.
You can consider this issue closed.

> Hi @myownhierarchy, sorry for the late reply. Did you managed to find a solution?
> 
> From the logs, it is indeed linked to gphoto, maybe your dslr has an auto-sleep/auto-off feature that makes it unavailable after some time?

closed 2024-02-03 by myownhierarchy