pibooth/pibooth #537: Tap 4 finger GUI

patt110, 2023-11-18:
The 4 finger tap does not work to open the GUI settings.
I still can use the esc button on the keyboard but not the 4 finger touch tap.

Any solutions?

Thanks in advance

patt110, 2023-11-20:
The menu automatically scrolls through the settings when I open the settings via esc.

System Bullseye
Camera: EOS600D
Touchscreen 15.6 with 10 Multitouchpoints

omgsideburns, 2024-01-02:
Are you still having these issues?  The four finger tap is more of a four finger hold, I find I have to hold it for about two seconds for the menu to pop up.

patt110, 2024-01-03:
Hi, yes the error is still available. 
I can hold the 4 Finger touch for more then two seconds and nothing will happen. If I open the menue about the keyboard ESC its automatically scroll through the menue. Could it be a problem with bullseye?

werdeil, 2024-01-29:
Hi @patt110, sorry for the late answer. The code finding the 4 finger touch is quite simple so my best guess would be that `pygame` may not see multiple keydowns on your screen. 
Except from this event we don't have other event that needs more than one finger on Pibooth so the rest of the app will work (but we don't have anywhere else to test my theory.

patt110, 2024-02-11:
I solve the issue, the problem was bullseye, I installed the buster version and it runs perfectly.

closed 2024-02-11 by patt110