pibooth/pibooth #538: Support for printing to Instax Link

sidharthb0, 2023-11-26:
**Describe the solution you'd like**
Wouldn't it be awesome to have your PiBooth print on original polaroids for that fully immersive Photo Booth experience? Polaroid's link printers are so freaking awesome and just the right device to bake into this setup. I understand CUPS doesn't work with bluetooth but I've identified an repo that allows the Polaroid Link printers to be connected to a raspberry pi.

[https://github.com/javl/InstaxBLE](url) Props to @javl for this awesome library.

I've tried to build a custom plugin to make this work with PiBooth but I am at my wits end. ChatGPT couldn't figure it out either. I was wondering if anyone could help build this (also requires disabling CUPS but still showing the print option after taking a photo). I am happy to contribute!

Cheers!

sidharthb0, 2023-12-11:
Managed to get this to work by building a custom plugin. Anyone interested can ping me; not perfect so not uploading it yet.

closed 2023-12-11 by sidharthb0

m-rousse, 2024-09-07:
Hi @sidharthb0, I'd be interested in using your custom plugin. Would it be possible to get access?

sidharthb0, 2024-09-07:
> Hi @sidharthb0, I'd be interested in using your custom plugin. Would it be possible to get access?

Hi @m-rousse, I've uploaded the instax plugin that I built here. It's been a while since I worked on this so I'm not exactly sure what other changes I had made (if any) in the pibooth code. Let me know if you're able to use it and if not I'll dig into it when I have some time and figure it out.
[pibooth_instax.py.zip](https://github.com/user-attachments/files/16917116/pibooth_instax.py.zip)

m-rousse, 2024-09-13:
Many thanks @sidharthb0 for the code + the instructions.
I managed to get it running and if I believe the logs, it should print just fine, I only need a cartridge to test now.

I had to do some tweaks on the path of the processed image, I'll see to share the diff of my changes.