pibooth/pibooth #540: Support non-legacy Raspberry Pi OS

jneuhauser, 2023-12-13:
**Is your feature request related to a problem? Please describe.**
Buster is end of life on June 30th, 2024.

**Describe the solution you'd like**
Support Bookworm and Bullseye

**Describe alternatives you've considered**
Running Buster until June 30th, 2024.

How about to track the required steps here in this ticket?

nsty, 2024-01-11:
Did you try it? I'm pretty sure I had it running on Bookworm without modification. I think the docs are just outdated.

omgsideburns, 2024-01-11:
I don't believe the current build release (2.0.8?) is working on bookworm as it's still utilizing the old camera libraries. 

I haven't dug through the 3.0 code to see if they've been implementing the new libraries yet either.

nsty, 2024-01-12:
You're right, it wasn't Bookworm. I just took a look. My photo booth was on Bullseye with Pibooth release 2.0.8. So this was definitively running without a hassle. Still, the issue is valid as Bookworm is the current release.

werdeil, 2024-06-15:
As written in https://github.com/pibooth/pibooth/issues/586#issuecomment-2169176932

> Debian 12 and all Python 3.11+ OS are indeed introducing a change on the way Python projects are installed by encouraging the use of virtualenv. We will need to update the install instruction to add something about it (either use virtualenv or the --break-system-packages option)

rogusdev, 2024-12-31:
Is there any summary of the known blockers? In case someone was to attempt to resolve this on their own?