pibooth/pibooth #542: OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected

IroMain327, 2024-01-08:
Hello, I have installed PiBooth on a Raspberry Pi 4B running in 64-bit mode. The camera is a V2. I am having trouble getting PiBooth to work; there is always a camera issue. Here is the problem: 
marguerite@raspberrypi:~ $ pibooth
pygame 2.1.2 (SDL 2.26.5, Python 3.11.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Generate the translation file in '/home/marguerite/.config/pibooth/translations.cfg'
[ INFO    ] pibooth           : Generate the configuration file in '/home/marguerite/.config/pibooth/pibooth.cfg'
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 33, in <module>
    sys.exit(load_entry_point('pibooth==2.0.8', 'console_scripts', 'pibooth')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pibooth-2.0.8-py3.11.egg/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pibooth-2.0.8-py3.11.egg/pibooth/booth.py", line 131, in __init__
    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy-1.3.0-py3.11.egg/pluggy/_hooks.py", line 493, in __call__
    return self._hookexec(self.name, self._hookimpls, kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy-1.3.0-py3.11.egg/pluggy/_manager.py", line 115, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy-1.3.0-py3.11.egg/pluggy/_manager.py", line 457, in traced_hookexec
    return outcome.get_result()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy-1.3.0-py3.11.egg/pluggy/_result.py", line 114, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/usr/local/lib/python3.11/dist-packages/pluggy-1.3.0-py3.11.egg/pluggy/_result.py", line 76, in from_call
    result = func()
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy-1.3.0-py3.11.egg/pluggy/_manager.py", line 454, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy-1.3.0-py3.11.egg/pluggy/_callers.py", line 130, in _multicall
    teardown[0].send(outcome)
  File "/usr/local/lib/python3.11/dist-packages/pibooth-2.0.8-py3.11.egg/pibooth/plugins/camera_plugin.py", line 28, in pibooth_setup_camera
    cam = camera.find_camera()
          ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pibooth-2.0.8-py3.11.egg/pibooth/camera/__init__.py", line 52, in find_camera
    raise EnvironmentError("Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected")
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected

somone help me ?

agrovista, 2024-01-09:
are you running buster os ?

werdeil, 2024-01-29:
Closing this issue as the support of new OS is followed by #540

closed 2024-01-29 by werdeil