pibooth/pibooth #544: "Oops! Something..."-error happens when light changes

Refurb, 2024-01-22:
**Describe the bug**
Hi there! Pibooth is amazing and I've used it now at some events at home and with friends. However, everytime I got the error "Oops, something went wrong error" for at least one time. I now notice this happens when something happens in the lightning of the room. For example: people are waving with light glowsticks, some discolights or lasers are in the composition of the lens. The camera can't handle that because I think it doesn't know where to focus on. When this happens, the Pibooth doesn't wait for a better moment to just make the picture. Nope: it gives the error. Whenever people get impatient it also happens the entire Pibooth software gets stuck and freezes and I have to turn off the power of the Raspberry.

Well, dear topic starter, don't you think this has anything to do with your DSLR camera? Well no, because whenever I run pibooth-diag command and use my flashlight in the lens it always succeeds to make a picture. So that's why I think this has something to do with the pibooth software. Thank you in advance for any answers. Read more here:

**Steps to reproduce the bug**
1. Start a new capture
2. Use the flash of your cellphone, point the light at the lens and wave it with fast movements to simulate the changing of lights in the space. 
3. You'll get an error. 

**Environment**
 - OS Raspbian GNU/Linux 11 (bullseye)
 - Camera  Canon EOS 1500D
 - Pibooth version: 2.0.7

**Log file pibooth**
`
[ INFO    ] pibooth           : Installed plugins: extra-lights-1.0.1, picture-template-1.1.0
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ DEBUG   ] pibooth           : Found gPhoto2 cameras on ports: 'usb:001,004'
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
[ DEBUG   ] pibooth           : Getting option actions/viewfinder=0
[ DEBUG   ] pibooth           : Setting option imgsettings/iso=100
[ WARNING ] pibooth           : Invalid value '100' for option iso (possible choices: ['Auto']), trying to set it anyway
[ DEBUG   ] pibooth.gphoto2   : ptp_usb_getresp [usb.c:515]: PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
[ DEBUG   ] pibooth.gphoto2   : _set_config [config.c:10780]: 'ptp_canon_eos_setdevicepropvalue (params, cursub->propid, &propval, cursub->type)' failed: PTP Device Busy (0x2019)
[ DEBUG   ] pibooth.gphoto2   : gp_context_error: The property 'ISO Speed' / 0xd103 was not set (0x2019: PTP Device Busy).
[ DEBUG   ] pibooth.gphoto2   : gp_camera_set_config [gphoto2-camera.c:1114]: 'camera->functions->set_config (camera, window, context)' failed: -110
[ ERROR   ] pibooth           : Unsupported option imgsettings/iso=100 ([-110] I/O in progress), configure your DSLR manually
[ DEBUG   ] pibooth           : Setting option settings/capturetarget=Memory card
[ INFO    ] pibooth           : Connected to printer 'Dai_Nippon_Printing_DS-RX1'
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'zTXt' 41 750
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 803 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 1203 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 1203 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 1221 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 1242 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 1242 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 1261 8192
[ DEBUG   ] pibooth           : took 7.536 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (4 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'zTXt' 41 750
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 803 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 1203 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 1203 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 1221 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 1242 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 1242 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 1261 8192
[ DEBUG   ] pibooth           : took 4.161 seconds
[... 965 lines cut ...]
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Gphoto2: b'ptp_usb_getresp [usb.c:515]': b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
Setting option actions/viewfinder="0"
Take a capture
Download file from DSLR
Save capture locally from memory buffer

================================================================================
SUCCESS : diagnostic completed
If you are investigating why pibooth does not work with your DSLR camera,
please paste the content of generated file 'diagnostic.log'
on https://github.com/pibooth/pibooth/issues

`

werdeil, 2024-01-26:
Hi @Refurb, thanks for the complete issue.

> The camera can't handle that because I think it doesn't know where to focus on. When this happens, the Pibooth doesn't wait for a better moment to just make the picture. Nope: it gives the error.

I think you've completely understood the issue, and on Pibooth the only thing that is done is to ask for a capture to Gphoto. If Gphoto cannot answer then we get an error.

However the fact that you didn't get the issue with pibooth-diag is quite strange as we are using the exact same command to gphoto. Are you using the Pibooth in hybrid mode (DSLR + picamera) or DSLR only?

Refurb, 2024-01-29:
Hey @werdeil , thank you for your reply.
I'm not using Picamera. I'm using the pibooth with a touchscreen and the Canon DSLR.
Did you test it as well by using the flashlight of your phone and making fast movements in front of the DSLR-camera?

werdeil, 2024-01-29:
If you are using only a DSLR then it is what I thought: in this configuration during the preview Pibooth is taking pictures in loop so if for one of the photo Gphoto is returning an issue then it will reach the "Oops" state.

There is nothing much we can do about that except improving the behavior in case of fail on Gphoto side. Can you describe again the current behavior? I'll try to see if I can find some improvements.

Refurb, 2024-01-30:
Thanks for your reaction, @werdeil. I now understand that the live preview are just a lot of photos and whenever the light changes in this preview, the Oops-state will be reached.

So an option is to disable the live preview? Not that I want that, but then there won't be an Oops-state?
And do I understand it correctly that you can also use the picamera which will only be used for the live preview while making the pictures? And then the Oops-problem will be gone?

Maybe you have to check the process of taking pictures in a loop. Can't you use live video of the camera? How are the ISO-settings? Maybe a lower frequency of photo's? Then you won't have the idea of a live preview, but you can do this of course:

Show a photo of the people in front of the camera - A black background with the number 3 (countdown) - photo again - Number 2 - photo again - Number 1 > after this the final photos will be made. 

The current behavior is listed in my first post: https://github.com/pibooth/pibooth/issues/544#issue-2094314733

Refurb, 2024-01-30:
Thanks for your comment and for shortening my logs @werdeil  

Do you have some more information/documentation about this hybrid setting? Or if I just plug the PiCamera into the Raspberry, Pibooth will know and it will use only the Picamera for the live preview? And where can I change the focal settings?

> Whenever people get impatient it also happens the entire Pibooth software gets stuck and freezes and I have to turn off the power of the Raspberry.

To clear this a bit out: I never had it but a friend which whom I made the photobooth with used it at his birthday. He told me that the software freezes. I thought this was a reaction to the  Oops-error being caused by the light changes in the live preview. Maybe it was something else. Let's just forget about this freeze and focus on the Oops-error being caused by a lot of changing lights in the live preview. 

Think I'm gonna give the PiCam a try. Let me know!

werdeil, 2024-01-30:
>if I just plug the PiCamera into the Raspberry, Pibooth will know and it will use only the Picamera for the live preview?

Yes we made it plug and play like, at startup pibooth will detected the device plugged and choose the corresponding setting.

> And where can I change the focal settings?

What I was meaning is that you need to setup your DSLR zoom to match the zoom level of the picamera so the preview is not showing more/less of the take image.

> Think I'm gonna give the PiCam a try. Let me know!

Take a picamera v2, we are not yet compatible with the v3 (see #526)

Refurb, 2024-01-31:
Thanks again @werdeil 

I've ordered the picamera v2 to try if this solves the problem. Let me know if you have solved the case.

Another small question: the raspberry is in a photobooth which is closed with screws. I want to check the camera device for media (the original pictures). Whenever the Raspberry starts, I see my Canon as a device on the desktop. Normally you can see the media in the explorer of Raspberry. However, as soon Pibooth runs the device is hidden. I've tried a lot of commands like killing pibooth and gphoto but nothing happens. I just want to see and delete the pictures on the DSLR. What can I do to release the use of gphoto/pibooth?

Refurb, 2024-02-02:
hi there! I've got the Raspberry Pi camera v2. After some testing it seems pretty okay to use!
How can I adjust the zoom of the Pi camera as you said, @werdeil ? Thanks!

werdeil, 2024-02-03:
On our setup it is done manually as we have a 18-55mm objective on it so we turned the objective ring to match the focal of the preview. What objective do you use on your Canon?

Refurb, 2024-02-03:
Hi @werdeil thanks for replying that fast.
I have a canon lens ef-s 24mm 1:2.8STMG 52mm Ø

Refurb, 2024-02-03:
Here you can see the difference. Left is the Picam preview, right is the Canon DSLR image. Both images were made after a command in the terminal window. So not within Pibooth.

I put a copy of the DSLR image over the PiCam-image. You can see the rectangle around it, so you have an idea about the proportion.

[image]

werdeil, 2024-02-03:
Oh I assumed (wrongly) that you had a zoom on the objective, if you are using a fixed focal it will be difficult to use both sorry.

Refurb, 2024-02-03:
Hmm. I'm going to try if I can crop the right and top border a bit. Which .py-script pibooth uses for the preview can I edit?