pibooth/pibooth #545: First image NIKON D5300 takes a long time after capture but before processing

djimkanters, 2024-01-25:
**Describe the bug**
The app works fine, however for the first image taken when pibooth is started the capture event get stuck on " SMILE " for ~30s. 

**Steps to reproduce the bug**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error.

**Logs**
pibooth
/usr/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected. Consider enabling compile time detection with environment variables like PYGAME_DETECT_AVX2=1 if you are compiling without cross compilation.
  return f(*args, **kwds)
pygame 2.5.2 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: picture-template-1.1.0, sftp-upload-1.0.0, printer_refill_message-0.1.0
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
AttributeError: 'str' object has no attribute 'decode'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
[... 518 lines cut ...]
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 4494 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 4512 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 4533 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 4533 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 4552 8192
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'zTXt' 41 6641
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 6694 390
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 7096 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 7096 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 7114 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 7135 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 7135 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 7154 8192

[diagnostic.log](https://github.com/pibooth/pibooth/files/14052229/diagnostic.log)

4. Post the `diagnostic.log` file if you are using a DSLR camera (generated by running the command `pibooth-diag`)

**Environment**
 Raspbian Buster
 NIKON D5300
 -Pibooth 2.08

werdeil, 2024-01-26:
Hi @djimkanters

From what I can see on the log seems that the camera is busy at the beginning. This is on Gphoto2 side and not Pibooth, I don't see what we could do on Pibooth side.
Do you confirm that this happens only on the first photo? 

Another possibility could be the autofocus, is your camera set on manual focus?

djimkanters, 2024-01-27:
Thanks, thats what i expected. It is indeed only the first image. Any suggestions on how to improve on the gphoto side? 

The camera is on manual focus.

closed 2024-01-27 by djimkanters