pibooth/pibooth #546: OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected

turriert, 2024-01-26:
Hello,

I try pibooth with :
- Raspberry Pi 4 B
- HDMI touch display 7" [Waveshare](https://www.waveshare.com/wiki/7inch_HDMI_LCD_(H)_(with_case))
- Pi camera v3 module

All is ok with libcamera-hello :
> pi@photobooth:~ $ libcamera-hello 
Made X/EGL preview window
[0:02:35.922445700] [1341]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:02:36.044667884] [1342]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media0 and ISP device /dev/media1
[0:02:36.044762551] [1342]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:02:36.045690601] [1341]  INFO Camera camera.cpp:1033 configuring streams: (0) 2304x1296-YUV420
[0:02:36.046181505] [1342]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA

but when i try with pibooth i have this :
> pi@photobooth:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py:469: PluggyTeardownRaisedWarning: A plugin raised an exception during an old-style hookwrapper teardown.
Plugin: pibooth-core:camera, Hook: pibooth_setup_camera
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected
For more information see https://pluggy.readthedocs.io/en/stable/api_reference.html#pluggy.PluggyTeardownRaisedWarning
  outcome = Result.from_call(
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 8, in <module>
    sys.exit(main())
File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 131, in __init__
    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 501, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 119, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 473, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 99, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 61, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 470, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 155, in _multicall
    teardown[0].send(outcome)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 28, in pibooth_setup_camera
    cam = camera.find_camera()
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/__init__.py", line 52, in find_camera
    raise EnvironmentError("Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected")
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected

Can you help me please !!

agrovista, 2024-01-26:
Need to run buster for the os and the older version of the pi cam (v2)
https://pibooth.readthedocs.io/en/stable/sources/install.html#hardware
https://pibooth.readthedocs.io/en/stable/sources/install.html#software

turriert, 2024-01-26:
Thank you.
When is the update due for bullseye and module 3 ?

werdeil, 2024-01-26:
Thanks @agrovista for the quick answer.

@turriert Unfortunately we don't have an estimate date for the compatibility with camera V3 and new OS, there is a lot of changes to do and we are not as available as we would like.

werdeil, 2024-01-29:
Closing this issue as the support of new os is followed by #540 and of picamera v3 (if not directly supported the new OS) by #526

closed 2024-01-29 by werdeil